wpilibsuite/allwpilib #1642: Unplugged XBox Controller Causes Weird Behavior on the dpad

rselwyn, 2019-03-16:
I'm not sure if this is an issue that other teams have faced, but we are finding that an unplugged Xbox controller has some undefined behavior going on with the dpad.  When we try to read from an unplugged xbox dpad, we are finding that the "bottom" button of the dpad always reads as true.

PeterJohnson, 2019-03-16:
Yep, DriverStation.getStickPOV() looks like it returns 0 in the case of either an unavailable POV or one that's not been initialized, when it should return -1.

synthmusic, 2019-03-18:
Just jumped on to report finding this as well. Adding a bump for priority as this is potentially dangerous: We extended Button based on POV position so we could associate Commands to dpad presses, In our case to drivetrain commands. The students wanted to test a command on the co-pilot controls, so didn't plug in the driver - and on enable the robot took off unexpectedly. 

You likely already realized this, but FWIW, it seems a previous strategy of checking if a POV request was to an index beyond the indexes for a given port would throw an exception - but the current DriverStation appears to instantiate all potential POVs for all potential Joysticks from the get go.

https://github.com/wpilibsuite/allwpilib/blob/a2368a6199b1ff5e22674c817e39c7431669f3d1/wpilibj/src/main/java/edu/wpi/first/wpilibj/DriverStation.java#L460-L465

https://github.com/wpilibsuite/allwpilib/blob/a2368a6199b1ff5e22674c817e39c7431669f3d1/wpilibj/src/main/java/edu/wpi/first/wpilibj/DriverStation.java#L199-L202

So the actual call goes through to 
https://github.com/wpilibsuite/allwpilib/blob/aab4c494d6f291a119adb9a00e15b7e081a82169/hal/src/main/java/edu/wpi/first/hal/HAL.java#L100

and into the native abyss....
https://github.com/wpilibsuite/allwpilib/blob/aab4c494d6f291a119adb9a00e15b7e081a82169/hal/src/main/native/cpp/jni/HAL.cpp#L194-L222

ThadHouse, 2019-03-18:
The JNIEnv is passed by the jni interface, and the Hal code that reads the joystick from the networking library is in the FRCDriverStation file in the Hal folder.

PeterJohnson, 2019-03-18:
The DriverStation Line 202 initializes the arrays, but the m_count of HALJoystickPOVs is initialized to 0 (Line 53).  Thus line 460 will report a warning that it's unplugged, *but* this just prints a message, and does not throw an exception (and we don't want it to; we don't want to crash the entire robot program just because a joystick is unplugged).  The issue is that function then proceeds on line 472 to return m_povs[pov], which is initialized to 0 (down).  It should instead return -1 circa line 465 to indicate the POV is not pressed.

NickRuisi, 2019-03-24:
We were impacted by this one yesterday. The arm on our robot is actuated by the D-pad and moved unexpectedly and very rapidly when we enabled our he robot. We were VERY lucky no one got hurt. This bug is a bad one due to risk of injury.

Kevin-OConnor, 2019-03-30:
I will re-iterate here what I have in other threads. If your robot moving rapidly when enabled leaves you feeling "lucky no one got hurt" then you should re-evaluate your safety practices. All robots should be assumed capable of any motion at any time while enabled. Software bugs happen, in both our code and team code. Sensors come unplugged sending PID loops wild. Any situation short of the robot not stopping when Disabled/E-Stopped should not result in an unsafe condition.

PR #1852 (2019-08-26): Correctly report -1 for POV on disconnected joystick
Fixes #1642.

C++ behavior was correct.
closed 2019-08-26 by PeterJohnson