wpilibsuite/allwpilib #1644: Failed to load networktables.ini during match

devjason, 2019-03-19:
The startup sequence failed to correctly load the networktables.ini file prior to starting robot code during a competition match, resulting in the program using default values specified in code.

The sequence of log files with relevant events looked like:
```
3/16/2019

15:27:29    Elimination - 20:1  135     - no activity
15:29:59    Elimination - 19:2  383     - match played
15:37:52    fn check            67      - system function checks performed, prefs OK
15:38:59    ?                   10      - see program start, no code/player robot activity
15:40:55    Elimination - 19:2  314     - no code/player robot activity
                                        - 15:45:15.223 we see "Code Start Notification"
                                        - 15:45:15.456 In Data Event Viewer see the following error message:
                                            NT: WARNING /home/lvuser/networktables.ini:1:header line mismatch, ignoring rest of file (Dispatcher.cpp:142)
                                            NT: WARNING: When reading initial persistent values from '/home/lvuser/networktables.ini':(Dispatcher.cpp:140)
                                        - Next line is "Robot program starting" 
15:46:28    Elimination - 20:1  320     - normal match start, but Prefs have not been loaded and robot fails to function.  
```

Please let me know if attaching a particular log file(s) would assist in determining the root cause of the issue.

PeterJohnson, 2019-03-19:
As the warning stated, it couldn't read the networktables.ini as the first line of that file didn't match what it was expecting to see.  My guess is the networktables.ini file was corrupted.  Was the robot rebooted/power cycled between the fn check and match?

devjason, 2019-03-19:
The code was rebooted via Driver Station at the end of the function check, which started the 10 second log.  At the end of the 10 second log is when the drivers untethered the DS.

We perform a write operation during RobotInit where we store an actuator position, but our other write operations are guarded by ContainsKey checks before writing new values. The reboot happened ~5 seconds after that write. We are using a Limelight this year, but I don't think they are writing any persistent NT entries. I just reviewed the codebase and I don't see any errant Put* operations.

Based on your description it sounds like the reboot at the end of the functional check is the most likely time any write corruption happened, but we should have been clear of any disk write operations at that time.  Once we get our test Rio unpacked I can watch the .ini file to verify it is not getting written.

PeterJohnson, 2019-03-19:
It may well be that there was a persistent value you didn't know about that was being regularly written, but only the original .ini file would indicate that, and it's probably been lost.  Anything loaded from the .ini file is automatically treated as being persistent, so if at any point in time a value was made persistent (even if by accident), it would be saved and reloaded as being persistent...

closed 2020-08-26 by calcmogul