wpilibsuite/allwpilib #1645: Go through joystick code, add things like connections, and maybe connection listeners

ThadHouse, 2019-03-20:
Things like the POV issue need to be looked at, away with giving users a way to check if connected or not. The name can in most cases be used to do this, but a better way of doing this would be better, along with potentially adding connection listeners.

Starlight220, 2020-04-15:
Define "connection listeners".

calcmogul, 2020-04-15:
User-defined callbacks for connect/disconnect events.

Starlight220, 2020-04-16:
> Things like the POV issue

What POV issue? I can fix most Java stuff, I just need to know exactly what is needed.

auscompgeek, 2020-04-16:
> What POV issue?

#1642

Starlight220, 2020-04-16:
Seems that it was fixed, though. 
What else needs to be done? 
And how should connection listeners be implemented? A thread that constantly runs seems like a bad idea to me.

Starlight220, 2023-09-28:
`GenericHID.isConnected()` exists for checking HID connection, and there are enough ways (in stdlibs and in wpilib) to poll a predicate that I think adding methods for connection callbacks (which are a niche use case) isn't worth it.
Can this issue be closed?