wpilibsuite/allwpilib #1646: Best way to approach getting a list of all cameras available

ThePfaffanater, 2019-03-23:
As far as I can tell there is no way to get a server from the Camera server instance without knowing its name. There also does not appear to be a way to actually get that name unless you already know about it. So the only two ways that appear to be feasible to support reading an unknown set of cameras client side is to have a separate network table of all the camera names the server side knows it created a server and read that client side and feed that into something like` CameraServer.getServer(<the server name>); ` TBH I never tested this idea so It may not even work. The other way I have seen is to read the CameraPublisher table and find all the servers with valid mjpeg stream links that I can create into separate HTTP cameras that exist client side. 
The former method is obviously a much more elegant way of doing this but I dont know if the CameraServer on the client's side has knowledge of all of the streams broadcasting from the server's (robot) side. From the documentation it does not seem like it does. And if it did I would see no reason for their not to be a 'getAllServers()' accessor. Since I am assuming you guys would not overlook that if it was possible, Im also assuming that CameraServer is only meant to be used from a server side perspective and I am just doing this all wrong.

Any help or clarifications are would be helpful here.

PeterJohnson, 2019-03-23:
Yes, as the name suggests, CameraServer is primarily intended to be used from the code that is "serving" (e.g. robot code), not in a dashboard client.  In particular, it has no functionality for creating a "global view" of camera names--the only server names in it are those created locally.

Enumerating the CameraPublisher NT table is the correct way to get a list of all published cameras from any source.

AustinShalit, 2019-08-09:
Please reopen if you have any more questions on this topic.

closed 2019-08-09 by AustinShalit