wpilibsuite/allwpilib #1649: Use of SPI port causes byte allocation errors

HiddenStorys, 2019-03-28:
The use of the SPI port appears to be causing byte allocation errors that are different from those generated when a memory overrun happens via user fault. When a memory overrun happens via user fault, the following error is generated: 
`There is insufficient memory for the Java Runtime Environment to continue. `
` Native memory allocation (mmap) failed to map x bytes for committing reserved memory. `
`  An error report file with more information is saved as: ...`
However, when the SPI memory overrun error happens, the following error occurs:
 ` terminate called without an active exception`
` FATAL: failed to allocate 4294967270 bytes`
(The number always begins with a four and is about that large, although the trailing digits do change.)
The reason I believe this is the SPI port is that, when the device using the SPI port is unplugged, the error never shows up. The timing of the 'failed to allocate x bytes' message always seems to be random.

ThadHouse, 2019-03-28:
What SPI device are you trying to use, and also can you post your code that uses SPI? That value is attempting to allocate basically the entire 32 bit address space, which seems very odd.

HiddenStorys, 2019-03-30:
We're trying to run a pixy2 off of the SPI port of the rio. We're using the drivers found here: https://github.com/HiddenStorys/Pixy2JavaAPI and the set up for the SPI link can be found here: https://github.com/HiddenStorys/Pixy2Vision/

AustinShalit, 2020-09-13:
I am closing this but if anyone has more info please leave a comment

closed 2020-09-13 by AustinShalit