wpilibsuite/allwpilib #1669: Replacement to SmartDashboard.putX

Daltz333, 2019-04-18:
Could we have a replacement method to quickly place and update information onto a specified tab?

Currently we can use `SmartDashboard.putNumber("Joystick X value", joystick1.getX());` but this always puts it onto the SmartDashboard tab. It'd be pretty useful to have a `tab.putNumber` which uses a tab instance. Or maybe even do `Shuffleboard.putNumber("name", value).withTab("TabName");` where it'll create the tab if not already exists. 

Why?
Because creating a widget instance for every single small debug variable is quite painful and not user intuitive at all.

SamCarlberg, 2019-04-18:
Seem #1608 would fix the underlying issue, though not in the way you're suggesting. It'd look something like

```java
Shuffleboard.getTab("Tab").addNumber("X", this::getX);
```
Instead of your suggestion:

```java
Shuffleboard.getTab("Tab").putNumber("X", getX());
```

I'd prefer to not have multiple slightly different approaches to doing the same thing, so it's one or the other. I prefer the lambda approach because it's a set-and-forget system; users won't need to remember to call the `putXYZ` method everywhere.

Daltz333, 2019-04-18:
Yeah. Didn't realize that issue was already open. Sounds good!

closed 2019-04-18 by Daltz333