wpilibsuite/allwpilib #1687: Failed to run a network table client-side program

tianer2820, 2019-05-16:
I followed the instruction on page https://wpilib.screenstepslive.com/s/currentCS/m/75361/l/851714-creating-a-client-side-program, but can't run the program after compiling. 

I copied the code as `networktablesdesktopclient/NetworkTablesDesktopClient.java`, and ran 

```
javac -cp wpiutil-java-3.0.0.jar:ntcore-java-4.0.0.jar:ntcore-jni-4.0.0-all.jar networktablesdesktopclient/NetworkTablesDesktopClient.java 
```

It does not throw any compile time error, but when I run it,
`java networktablesdesktopclient.NetworkTablesDesktopClient`,
 it throws an exception: 

```
Exception in thread "main" java.lang.NoClassDefFoundError: edu/wpi/first/networktables/NetworkTableInstance
	at networktablesdesktopclient.NetworkTablesDesktopClient.run(NetworkTablesDesktopClient.java:12)
	at networktablesdesktopclient.NetworkTablesDesktopClient.main(NetworkTablesDesktopClient.java:8) 
```

I'm not sure this is a bug or I'm doing something wrong.

Here's the code I used:
```
package networktablesdesktopclient;
import edu.wpi.first.networktables.NetworkTable;
import edu.wpi.first.networktables.NetworkTableEntry;
import edu.wpi.first.networktables.NetworkTableInstance;

public class NetworkTablesDesktopClient {
  public static void main(String[] args) {
    new NetworkTablesDesktopClient().run();
  }

  public void run() {
    NetworkTableInstance inst = NetworkTableInstance.getDefault();
    NetworkTable table = inst.getTable("datatable");
    NetworkTableEntry xEntry = table.getEntry("x");
    NetworkTableEntry yEntry = table.getEntry("y");
    inst.startClientTeam(1234);  // where TEAM=190, 294, etc, or use inst.startClient("hostname") or similar
    while (true) {
      try {
        Thread.sleep(1000);
      } catch (InterruptedException ex) {
        System.out.println("interrupted");
        return;
      }
      double x = xEntry.getDouble(0.0);
      double y = yEntry.getDouble(0.0);
      System.out.println("X: " + x + " Y: " + y);
    }
  }
}
```

ThadHouse, 2019-05-16:
You need to pass the -cp arguments to the `java` command, not just the `javac` command. Otherwise the java execution cannot find the library jars.

tianer2820, 2019-05-16:
@ThadHouse thanks, it works when I run this:
```
java -cp wpiutil-java-3.0.0.jar:ntcore-java-4.0.0.jar:ntcore-jni-4.0.0-all.jar:./ networktablesdesktopclient.NetworkTablesDesktopClient
```

closed 2019-05-16 by tianer2820