wpilibsuite/allwpilib #1692: Add unit tests for MotorSafety class

calcmogul, 2019-05-17:
There currently are none. We had an incident last year with MotorSafety breaking right before kickoff, so we should restructure MotorSafety to be more testable, then actually add unit tests.

Starlight220, 2020-04-19:
I can fix Java. What is needed?

Starlight220, 2020-04-19:
Now that I think of it, MotorSafety can be refactored to run with a Watchdog instance. It already implicitly uses a Watchdog mechanism, it should explicitly use a Watchdog instance.

calcmogul, 2020-04-19:
We already tried that. That's what I was referring to with "MotorSafety breaking right before kickoff". It got reverted.

calcmogul, 2020-04-19:
The unit tests need to test all the code paths. Namely, that all the public API functions work as advertised and that expiration actually occurs and stops the motor when it's supposed to.

Starlight220, 2020-04-19:
> > MotorSafety can be refactored to run with a Watchdog instance
> 
> We already tried that. That's what I was referring to with "MotorSafety breaking right before kickoff". It got reverted.

Are we planning on trying again, or are we leaving it as is?

Starlight220, 2020-04-19:
I will now start working on tests.

Starlight220, 2020-04-19:
How am I supposed to test an abstract class? I can't create an instance of it...

AustinShalit, 2020-04-19:
You can create a testing implementation with some basic functionality.

Starlight220, 2020-04-19:
Is there an example somewhere?

calcmogul, 2020-04-19:
You make a mock class that implements MotorSafety.