wpilibsuite/allwpilib #1700: Add H-Drive to Drive options.

MC42, 2019-05-28:
WPILib currently has options for tank / WCD (differentialdrive), Kiwi drive (KilloughDrive), mecanum / x drive (MecanumDrive), but nothing for H-drives specifically.  A helper class to make the controlling and construction of H-Drive would seemingly help teams drive them more effectively, and avoid having to [suggest wading through other teams code](https://www.chiefdelphi.com/t/idiots-guide-to-programming/358555?u=tjf) to understand.

AustinShalit, 2019-05-29:
What way do you think teams mount their center motor more often, shaft pointing to the front of the robot or to the back of the robot (of course this will be an option; just looking for a default)?

MC42, 2019-05-29:
I personally think it'd be strafe-motor-shaft front-facing, given that's what the three designs I've seen for them have done.

calcmogul, 2019-05-29:
For mathematical purity, I'd do positive input results in movement along the positive y axis. Teams are always having to invert random things in software to make the robot behave anyway.

calcmogul, 2023-11-21:
OBE by FRC 2024 drivetrain rules: https://www.firstinspires.org/robotics/frc/blog/2023-motor-rule-changes-for-the-2024-season

closed 2023-11-21 by calcmogul