wpilibsuite/allwpilib #1701: Add SwerveDrive class

calcmogul, 2019-05-28:
Once the modularized PIDController and inverse kinematics classes land, this should be pretty easy to put together. I suggest a SwerveDrive class for inverse kinematics returning four pairs of wheel speeds and angles for a list of four locations provided in the constructor. An example plumbing it together is a must.

SamCarlberg, 2019-05-29:
To my knowledge, there are two basic types of swerve drives: one where all wheels are chained together, and one where every wheel can rotate independently. How do you intend to make that distinction? For example, a corotating configuration can only turn in place with skid-steer, but a configuration with independent modules can spin in place by arranging the wheels in a diamond pattern.

calcmogul, 2019-05-29:
I've heard that called crab drive.

prateekma, 2019-09-30:
@calcmogul I think this can be closed now

closed 2019-09-30 by calcmogul