wpilibsuite/allwpilib #1705: Missing smartdashboard.vbs and shuffleboard.vbs

L1391, 2019-06-06:
So I am just trying to install the Smart Dashboard and Shuffle Board from the WPILib Suite. I get the desktop shortcuts but the frc2019/tools folder does not have the vbs files. 

I checked Java/JDK and Tools on the install program. Is that all I need?

I have a Windows 10 64bit Dell computer

ThadHouse, 2019-06-06:
Hmm I should double check that just selecting those 2 is enough. Because someone else reported this too. 

But either way you should have a tools updater jar in that folder. If you run that, it will create the vbs's

L1391, 2019-06-06:
Sorry I am not to technical, how do you run a JAR file? What java version do I need?

L1391, 2019-06-06:
Ok I installed the latest Java 8 and tried running the JAR but got the following pop up

Error: A JNI error has occurred, please check your installation and try again

ThadHouse, 2019-06-06:
It needs Java 11. Java 11 is installed with the wpilib tools, and can be found in the frc2019/jdk folder.

L1391, 2019-06-06:
The jdk folder has a lot in it. How do I install Java 11 or ensure that the JAR I run by Java 11?

ThadHouse, 2019-06-06:
Ok. In that tools folder, do you have a ToolsUpdater.bat? If so, run that. If not you can download it from here, place it in the tools folder and run it.

https://github.com/wpilibsuite/WPILibInstaller/blob/master/files/ToolsUpdater.bat

If that still doesn't work, its likely easiest to just manually add the vbs scripts. That is located at the location before. Just download it, and rename copies to `SmartDashboard.vbs` and `Shuffleboard.vbs`.  You don't need to change any of the contents of them, just the file name.

L1391, 2019-06-06:
Ok the batch file does not run. What is "the location before" and "it"?

ThadHouse, 2019-06-06:
I meant below, and the link is this
https://github.com/wpilibsuite/WPILibInstaller/blob/master/files/ScriptBase.vbs

How it the batch file failing though? If you run it from a terminal, does it give an error?

L1391, 2019-06-06:
Ok I did both

Batch file I think runs but nothing happens

The new vbs files give
Error: Unable to access jarfile

ThadHouse, 2019-06-06:
Ok. So that means youre likely missing the jars as well from the tools folder. Since that batch file won't run, that's usually caused by a system misconfiguration. I'm at work now so I can try and help later to get the jars added correctly too.

L1391, 2019-06-06:
Ok I will talk to you later then. The files in the tools folder are as follows:

Scriptbase.py
Scriptbase
(Copy of scriptbase) shuffleboard
(Copy of scriptbase) smartdashboard
tools.json
ToolsUpdater
ToolsUpdater.jar
ToolsUpdater.pu