wpilibsuite/allwpilib #1765: Run Java Integration Tests on roboRIO

AustinShalit, 2019-07-16:
The test suite that is run on the roboRIO should include all of the tests that are a part of wpilibj. 

Depends on the test stand getting fixed.

ThadHouse, 2019-07-16:
I won't be able to run all of them, as some of the wpilibj tests depend on the simulator. But there are some critical ones that do need to be ran, specially the preferences run.

AustinShalit, 2019-07-16:
I can tag them as simulator and then tell the rio to not run things tagged with `@Simulator`

calcmogul, 2021-07-11:
> The test suite that is run on the roboRIO should include all of the tests that are a part of wpilibj.

The consensus nowadays seems to be that the test bench should only run tests that exercise hardware; things that have adequate unit tests shouldn't be run on the RIO.

closed 2021-07-11 by calcmogul