wpilibsuite/allwpilib #1837: Are Shuffleboard component properties really whitespace-insensitive?

tylertian123, 2019-08-20:
[The docs](https://first.wpi.edu/FRC/roborio/release/docs/java/edu/wpi/first/wpilibj/shuffleboard/ShuffleboardComponent.html#withProperties(java.util.Map)) say that property names for `ShuffleboardComponent.withProperties(Map)` is case and whitespace insensitive:
> Property names are case- and whitespace-insensitive (capitalization and spaces do not matter).

If I understood this correctly, wouldn't it mean that `"Color when true"`, is equivalent to `"color when true"` and `"colorwhentrue"`? But when I tried to do this:
```java
Shuffleboard.getTab("TestTab").add("A Boolean", true)
        .withWidget(BuiltInWidgets.kBooleanBox)
        .withProperties(Map.of("colorwhentrue", "blue"));
```
The property is not set. Whereas using `"color when true"` or `"Color when true"` correctly sets the property. Am I misunderstanding what being “whitespace-insensitive” means, or is this a mistake in the documentation?

terbos90803, 2019-08-20:
whitespace-insensitive means that at the location in the syntax that requires whitespace, the type and amount of whitespace doesn't matter, as long as there is something to act as a delimiter between the other parts of the string.  The parser is not smart enough to figure out where there should be breaks between words without some whitespace to help it.

tylertian123, 2019-08-20:
Thanks for the clarification! I'm still confused about what you mean by "depending on the parser". Does WPILib not have its own parser for these things?

terbos90803, 2019-08-20:
I was speaking in general terms of how parsers handle whitespace as a delimiter. 
I have not looked at the internals of the specific parser in Shuffleboard, but it’s always safe to assume the space character will work as a delimiter in this type of string usage. 
It’s when you get to language compilers, the allowed whitespace types expand to include tabs and sometimes newlines. Newlines certainly depends on the language (line based vs not)

closed 2020-08-26 by calcmogul