wpilibsuite/allwpilib #1881: Building example project for 2020

jreneew2, 2019-09-08:
Hi,
I am trying to build the latest version of wpilib to just test out new features for 2020 because of the large number of changes coming next year that are relevant to us. I was able to successfully build the [toolchain](https://github.com/wpilibsuite/roborio-toolchain) on my linux machine (for windows). I then build using this command:
`.\gradlew.bat build -PtoolChainPath=C:'C:\Users\Drew Williams\Documents\wpilib_dev\frc2020\roborio\bin\'`

I run that command from the root of the cloned git repo and get this, however i dont think its a problem because it seems to build after it cant configure any of the subprojects...
`> Configure project :simulation:lowfi_simulation                                                                        INFO: Could not find files for the given pattern(s).                                                                    INFO: Could not find files for the given pattern(s).
INFO: Could not find files for the given pattern(s).`

So, now I have the built wpilib, how would I be able to use the library in an actual project? I see there are examples that are built, but I was wondering if I could include them in my own project?

Thanks!

ThadHouse, 2019-09-08:
The stuff for 2020 actually requires the 2020 roborio image. It will not work correctly with the 2019 image. We currently don't even have a copy of the 2020 image, and once we do only beta teams will have access to is. So it's not currently possible to use what you built on a real robot.

However, there is a jitpack way of getting the new command framework working in a 2019 project, I'll ping the person who knows how to do that so he can reply with that information.

That info message is just a message printed internally by Gradle, it can be completely ignored, it's not causing stuff to rebuild.

Also, you don't need to build your own toolchain, technically you can just run gradlew installRoborioToolchain and it will download and install our prebuilt ones. That toolchain path variable isn't needed, and actually doesn't even work anymore. It worked because you placed the compiler in the right path so Gradle knew to look for it.

jreneew2, 2019-09-08:
Oh I didn't realize that the newer releases were available with that command. I wasn't planning on running a real robot with the 2020 library, however, the 2019 port sounds interesting. 

Thanks for all the information!

closed 2019-09-08 by jreneew2