wpilibsuite/allwpilib #1887: Generic TCP port forwarder

PeterJohnson, 2019-09-14:
The USB and Ethernet networks are separate networks on the RoboRIO.  This means teams can't tether via USB and access anything on the Ethernet network.  CameraServer provides camera stream forwarding, but there's no feature currently to make it easy for teams to port-forward to things like a coprocessor webserver.  With the increasing use of coprocessors, a simple generic port forwarder for TCP connections would be a useful addition to the library.  Conceptual interface would be `add(rioPort, destAddress, destPort)` and `remove(rioPort)`.  The idea would be for just a raw TCP forwarder, not a full web proxy, so some things wouldn't work, but it would likely still be an improvement over the current situation.

virtuald, 2019-09-14:
Why not something that configures the linux kernel to bridge the two? Or something of that nature.

PeterJohnson, 2019-09-14:
That gets ugly fast, in part due to the way the USB interface works.  I doubt NI would want to implement it, and we wouldn't want to do it without their involvement.

PR #1890 (2019-09-27): Add PortForwarder class
This class provides an easy way to forward local ports to another host/port.
This is primarily useful to provide a way to access Ethernet-connected devices
from a computer tethered to the RoboRIO USB port.

The most natural spot to put the shared implementation of this class was into
wpiutil, so a wpiutilJNI library has been added.

Fixes #1887.
closed 2019-09-27 by PeterJohnson