wpilibsuite/allwpilib #1888: Add Joystick wrapper classes to make Command buttons easier to use

PeterJohnson, 2019-09-14:
As suggested by a user, this would allow replacing this:
```
XboxController driverController = new XboxController(0);
JoystickButton aButton = new JoystickButton(driverController, XboxController.Buttons.A.value);

aButton.whenPressed(new ExampleCommand());
```
with this:
```
XboxController driverController = new XboxController(0);

driverController.getAButton().whenPressed(new ExampleCommand());
```

Note this needs to be a wrapper class--we don't want to change the main XboxController and Joystick classes because of both dependencies and potential confusion of the meaning of getButton().

Given that, the question is naming; while it could be done with packages/namespaces, that seems like a recipe for getting the classes confused?  Is there a reasonable prefix that's not too ugly?

Oblarg, 2019-09-14:
I don't think this is necessary.  With a proper static import, the difference is actually between:

```java
XboxController driverController = new XboxController(0);
new JoystickButton(driverController, Buttons.A.value).whenPressed(new ExampleCommand());
```

and 

```java
XboxController driverController = new XboxController(0);

driverController.getAButton().whenPressed(new ExampleCommand());
```

That's not a very big difference in actual verbosity/complexity for the user.

Starlight220, 2020-04-20:
Is this settled? If so, should be closed.

Starlight220, 2020-04-27:
I know there isn't much difference with a proper static import, however: maybe add a ctor overload for JoystickButton that takes for the port number an `XBoxController.Buttons` value instead of an int? The `Buttons.XXX.value` is a bit annoying and verbose, I think adding the ctor overload would clean up code.

joshuathooyavan, 2021-12-20:
I can work on this, but do we want to continue with @Starlight220 's suggestion of taking in an `XboxController.Buttons` parameter? That way with a static import it can be as short as
`
new JoystickButton(driverController, kA).whenPressed(new ExampleCommand());
`
But if we're going for conciseness, why not just:
`
new Button(driverController::getAButton).whenPressed(new ExampleCommand());
`
This requires no changes and is the shortest way I can think of to write it.

Oblarg, 2021-12-20:
@joshuathooyavan you should PR that solution into one of the frc-docs articles; it is indeed pretty nice.

joshuathooyavan, 2021-12-23:
#3826 is my PR for this.
Need help with writing Java tests and also C++.

Bankst, 2021-12-23:
Per [discussion on Discord](https://discord.com/channels/176186766946992128/368993897495527424/922620115479060640), we landed on an API as such
```java
CommandXboxController xbCtrl = new CommandXboxController(1);
xbCtrl.leftBumper().whenPressed(new PrintCommand("Left Bumper Pressed"));
```

Starlight220, 2022-03-29:
Should be linked to #3826.

PR #4247 (2022-06-14): [wpilib+commands] BooleanEvent/Trigger factories on HID classes
Resolves #1888.
Closes #3826.

Depends on #4104.
closed 2022-06-14 by PeterJohnson