wpilibsuite/allwpilib #1932: Build Broken on MacOS 10.15

AustinShalit, 2019-10-10:
```
In file included from /Users/austinshalit/git/allwpilib/wpilibc/src/main/native/cpp/buttons/HeldButtonScheduler.cpp:10:
In file included from /Users/austinshalit/git/allwpilib/wpilibc/src/main/native/include/frc/buttons/Button.h:10:
/Users/austinshalit/git/allwpilib/wpilibc/src/main/native/include/frc/buttons/Trigger.h:37:3: error: explicitly defaulted move constructor is implicitly deleted [-Werror,-Wdefaulted-function-deleted]
  Trigger(Trigger&&) = default;
  ^
/Users/austinshalit/git/allwpilib/wpilibc/src/main/native/include/frc/buttons/Trigger.h:51:20: note: move constructor of 'Trigger' is implicitly deleted because field 'm_sendablePressed' has a deleted move constructor
  std::atomic_bool m_sendablePressed{false};
                   ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/atomic:1085:7: note: copy constructor of 'atomic<bool>' is implicitly deleted because base class '__atomic_base<bool>' has a deleted copy constructor
    : public __atomic_base<_Tp>
      ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/atomic:984:5: note: '__atomic_base' has been explicitly marked deleted here
    __atomic_base(const __atomic_base&) = delete;
    ^
In file included from /Users/austinshalit/git/allwpilib/wpilibc/src/main/native/cpp/buttons/HeldButtonScheduler.cpp:10:
In file included from /Users/austinshalit/git/allwpilib/wpilibc/src/main/native/include/frc/buttons/Button.h:10:
/Users/austinshalit/git/allwpilib/wpilibc/src/main/native/include/frc/buttons/Trigger.h:38:12: error: explicitly defaulted move assignment operator is implicitly deleted [-Werror,-Wdefaulted-function-deleted]
  Trigger& operator=(Trigger&&) = default;
           ^
/Users/austinshalit/git/allwpilib/wpilibc/src/main/native/include/frc/buttons/Trigger.h:51:20: note: move assignment operator of 'Trigger' is implicitly deleted because field 'm_sendablePressed' has a deleted move assignment operator
  std::atomic_bool m_sendablePressed{false};
                   ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/atomic:1085:7: note: copy assignment operator of 'atomic<bool>' is implicitly deleted because base class '__atomic_base<bool>' has a deleted copy assignment operator
    : public __atomic_base<_Tp>
      ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/atomic:985:20: note: 'operator=' has been explicitly marked deleted here
    __atomic_base& operator=(const __atomic_base&) = delete;
                   ^
In file included from /Users/austinshalit/git/allwpilib/wpilibc/src/main/native/cpp/buttons/HeldButtonScheduler.cpp:10:
/Users/austinshalit/git/allwpilib/wpilibc/src/main/native/include/frc/buttons/Button.h:29:3: error: explicitly defaulted move constructor is implicitly deleted [-Werror,-Wdefaulted-function-deleted]
  Button(Button&&) = default;
  ^
/Users/austinshalit/git/allwpilib/wpilibc/src/main/native/include/frc/buttons/Button.h:26:16: note: move constructor of 'Button' is implicitly deleted because base class 'frc::Trigger' has a deleted move constructor
class Button : public Trigger {
               ^
/Users/austinshalit/git/allwpilib/wpilibc/src/main/native/include/frc/buttons/Trigger.h:37:3: note: copy constructor is implicitly deleted because 'Trigger' has a user-declared move constructor
  Trigger(Trigger&&) = default;
  ^
In file included from /Users/austinshalit/git/allwpilib/wpilibc/src/main/native/cpp/buttons/HeldButtonScheduler.cpp:10:
/Users/austinshalit/git/allwpilib/wpilibc/src/main/native/include/frc/buttons/Button.h:30:11: error: explicitly defaulted move assignment operator is implicitly deleted [-Werror,-Wdefaulted-function-deleted]
  Button& operator=(Button&&) = default;
          ^
/Users/austinshalit/git/allwpilib/wpilibc/src/main/native/include/frc/buttons/Button.h:26:16: note: move assignment operator of 'Button' is implicitly deleted because base class 'frc::Trigger' has a deleted move assignment operator
class Button : public Trigger {
               ^
/Users/austinshalit/git/allwpilib/wpilibc/src/main/native/include/frc/buttons/Trigger.h:37:3: note: copy assignment operator is implicitly deleted because 'Trigger' has a user-declared move constructor
  Trigger(Trigger&&) = default;
  ^
4 errors generated.

In file included from /Users/austinshalit/git/allwpilib/wpilibc/src/main/native/cpp/buttons/Button.cpp:8:
In file included from /Users/austinshalit/git/allwpilib/wpilibc/src/main/native/include/frc/buttons/Button.h:10:
/Users/austinshalit/git/allwpilib/wpilibc/src/main/native/include/frc/buttons/Trigger.h:37:3: error: explicitly defaulted move constructor is implicitly deleted [-Werror,-Wdefaulted-function-deleted]
  Trigger(Trigger&&) = default;
  ^
/Users/austinshalit/git/allwpilib/wpilibc/src/main/native/include/frc/buttons/Trigger.h:51:20: note: move constructor of 'Trigger' is implicitly deleted because field 'm_sendablePressed' has a deleted move constructor
  std::atomic_bool m_sendablePressed{false};
                   ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/atomic:1085:7: note: copy constructor of 'atomic<bool>' is implicitly deleted because base class '__atomic_base<bool>' has a deleted copy constructor
    : public __atomic_base<_Tp>
      ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/atomic:984:5: note: '__atomic_base' has been explicitly marked deleted here
    __atomic_base(const __atomic_base&) = delete;
    ^
In file included from /Users/austinshalit/git/allwpilib/wpilibc/src/main/native/cpp/buttons/Button.cpp:8:
In file included from /Users/austinshalit/git/allwpilib/wpilibc/src/main/native/include/frc/buttons/Button.h:10:
/Users/austinshalit/git/allwpilib/wpilibc/src/main/native/include/frc/buttons/Trigger.h:38:12: error: explicitly defaulted move assignment operator is implicitly deleted [-Werror,-Wdefaulted-function-deleted]
  Trigger& operator=(Trigger&&) = default;
           ^
[... 540 lines cut ...]
#define WPI_DEPRECATED(msg) [[deprecated(msg)]]
                              ^
1 warning and 2 errors generated.

> Task :wpilibj:pmdMain
This analysis could be faster, please consider using Incremental Analysis: https://pmd.github.io/pmd-6.7.0/pmd_userdocs_incremental_analysis.html

> Task :wpilibc:compileWpilibcBaseOsxx86-64DebugStaticLibraryWpilibcBaseCpp
...output for 17 more failed operation(s) continued in file:///Users/austinshalit/git/allwpilib/wpilibc/build/tmp/compileWpilibcBaseOsxx86-64DebugStaticLibraryWpilibcBaseCpp/output.txt.

> Task :wpilibc:compileWpilibcBaseOsxx86-64DebugStaticLibraryWpilibcBaseCpp FAILED

> Task :wpilibc:compileWpilibcTestOsxx86-64DebugGoogleTestExeWpilibcTestCpp
In file included from /Users/austinshalit/git/allwpilib/wpilibc/src/test/native/cpp/frc2/command/WaitCommandTest.cpp:9:
/Users/austinshalit/git/allwpilib/wpilibc/src/main/native/include/frc2/command/WaitCommand.h:35:3: error: explicitly defaulted copy constructor is implicitly deleted [-Werror,-Wdefaulted-function-deleted]
  WaitCommand(const WaitCommand& other) = default;
  ^
/Users/austinshalit/git/allwpilib/wpilibc/src/main/native/include/frc2/command/WaitCommand.h:46:14: note: copy constructor of 'WaitCommand' is implicitly deleted because field 'm_timer' has a deleted copy constructor
  frc::Timer m_timer;
             ^
/Users/austinshalit/git/allwpilib/wpilibc/src/main/native/include/frc/Timer.h:60:3: note: copy constructor is implicitly deleted because 'Timer' has a user-declared move constructor
  Timer(Timer&& rhs);
  ^
1 error generated.
```

Oblarg, 2019-10-17:
Same error on 10.14.6.

PeterJohnson, 2019-10-21:
Fixed by #1954.

closed 2019-10-21 by PeterJohnson