wpilibsuite/allwpilib #1946: #include "edu_wpi_first_wpiutil_WPIUtilJNI.h"

uutzinger, 2019-10-17:
`allwpilib/wpiutil/src/main/native/cpp/jni/WPIUtilJNI.cpp`
is asking to include `edu_wpi_first_wpiutil_WPIUtilJNI.h`
which does not seem to exist on `https://github.com/wpilibsuite/allwpilib`

ThadHouse, 2019-10-17:
It's generated during the build, and not stored in repo.

Are you running into a build issue?

uutzinger, 2019-10-18:
I was working with FRCVision-pi-gen which uses robotpy-cscore for python bindings. robotpy-cscore has a setup.py which tries to build wpiutil and cscore and then runs into include issues.
I build allwpilib using 
```
cmake -D WITHOUT_ALLWPILIB=OFF \
    -D CMAKE_BUILD_TYPE=Release \
    -D OPENCV_JAR_FILE=`ls /usr/local/java/opencv-411.jar` \
    -D OPENCV_JNI_FILE=`ls /usr/local/lib/libopencv_java411.so` \
    -D OpenCV_DIR=/usr/local/share/opencv4 \
    -D THREADS_PTHREAD_ARG=-pthread \
    -D BUILD_SHARED_LIBS=ON \
    -D CMAKE_INSTALL_PREFIX=/usr/local ..
```
I am not sure if cmake approach does not create the include files.

ThadHouse, 2019-10-18:
Cmake definitely does create those files, and we build cmake in our CI, so it definitely does work. 

If you're just building py bindings, you likely could disable Java entirely with WITHOUT_JAVA=ON that would bypass any java things anyway, which are only needed for actual java interop.

PeterJohnson, 2019-10-19:
FRCVision-pi-gen doesn't use setup.py to build robotpy-cscore, it just builds the shared object manually.  What are you trying to do?

uutzinger, 2019-10-22:
I was following the FRCVision pi gen process but with intent to complete it on already existing raspi installation or on other platform. When I reached robotpy-python generation I tried the manual build but it seemed setup.py was the standard way.

Likely issue is fixed by not  executing  `make clean`  until anything else that might need the includes is completed (e.g. robotpy-cscore)

closed 2019-10-22 by uutzinger