wpilibsuite/allwpilib #1965: Command Based Template Doesn't Build On Mac

Kython89, 2019-10-21:
> Executing task: ./gradlew build generateVsCodeConfig   -Dorg.gradle.java.home="/Users/kyle/frc2019/jdk" <

Downloading https://services.gradle.org/distributions/gradle-5.4.1-bin.zip
...................................................................................

> Task :compileFrcUserProgramTestOsxx86-64DebugGoogleTestExeFrcUserProgramTestCpp FAILED
In file included from /Users/kyle/frc-2019-beta/frc-2019-beta/src/test/cpp/main.cpp:1:
In file included from /Users/kyle/.gradle/caches/transforms-2/files-2.1/a6db147c4a6c78e71e2ef2d4bc36f68e/hal-cpp-2020.1.1-beta-1-headers/hal/HAL.h:39:
In file included from /Users/kyle/.gradle/caches/transforms-2/files-2.1/a6db147c4a6c78e71e2ef2d4bc36f68e/hal-cpp-2020.1.1-beta-1-headers/hal/SimDevice.h:15:
In file included from /Users/kyle/.gradle/caches/transforms-2/files-2.1/813d3f1bfc80aadce9047e7e0690a758/wpiutil-cpp-2020.1.1-beta-1-headers/wpi/ArrayRef.h:15:
/Users/kyle/.gradle/caches/transforms-2/files-2.1/813d3f1bfc80aadce9047e7e0690a758/wpiutil-cpp-2020.1.1-beta-1-headers/wpi/STLExtras.h:34:10: fatal error: 'optional' file not found
#include <optional>
         ^~~~~~~~~~
1 error generated.

> Task :compileFrcUserProgramOsxx86-64DebugExecutableFrcUserProgramCpp FAILEDIn file included from /Users/kyle/frc-2019-beta/frc-2019-beta/src/main/cpp/subsystems/ExampleSubsystem.cpp:8:
In file included from /Users/kyle/frc-2019-beta/frc-2019-beta/src/main/include/subsystems/ExampleSubsystem.h:10:
In file included from /Users/kyle/.gradle/caches/transforms-2/files-2.1/6a64cf007cf067aa3f4c77f31619150a/wpilibc-cpp-2020.1.1-beta-1-headers/frc2/command/Subsy
stemBase.h:11:
In file included from /Users/kyle/.gradle/caches/transforms-2/files-2.1/6a64cf007cf067aa3f4c77f31619150a/wpilibc-cpp-2020.1.1-beta-1-headers/frc/smartdashboard/SendableHelper.h:13:
In file included from /Users/kyle/.gradle/caches/transforms-2/files-2.1/813d3f1bfc80aadce9047e7e0690a758/wpiutil-cpp-2020.1.1-beta-1-headers/wpi/Twine.h:14:
In file included from /Users/kyle/.gradle/caches/transforms-2/files-2.1/813d3f1bfc80aadce9047e7e0690a758/wpiutil-cpp-2020.1.1-beta-1-headers/wpi/StringRef.h:13:
/Users/kyle/.gradle/caches/transforms-2/files-2.1/813d3f1bfc80aadce9047e7e0690a758/wpiutil-cpp-2020.1.1-beta-1-headers/wpi/STLExtras.h:34:10: fatal error: 'optional' file not found
#include <optional>
         ^~~~~~~~~~
1 error generated.

In file included from /Users/kyle/frc-2019-beta/frc-2019-beta/src/main/cpp/RobotContainer.cpp:8:
In file included from /Users/kyle/frc-2019-beta/frc-2019-beta/src/main/include/RobotContainer.h:10:
In file included from /Users/kyle/.gradle/caches/transforms-2/files-2.1/6a64cf007cf067aa3f4c77f31619150a/wpilibc-cpp-2020.1.1-beta-1-headers/frc2/command/Command.h:10:
In file included from /Users/kyle/.gradle/caches/transforms-2/files-2.1/6a64cf007cf067aa3f4c77f31619150a/wpilibc-cpp-2020.1.1-beta-1-headers/frc/ErrorBase.h:12:
In file included from /Users/kyle/.gradle/caches/transforms-2/files-2.1/813d3f1bfc80aadce9047e7e0690a758/wpiutil-cpp-2020.1.1-beta-1-headers/wpi/StringRef.h:13:
/Users/kyle/.gradle/caches/transforms-2/files-2.1/813d3f1bfc80aadce9047e7e0690a758/wpiutil-cpp-2020.1.1-beta-1-headers/wpi/STLExtras.h:34:10: fatal error: 'optional' file not found
#include <optional>
         ^~~~~~~~~~
1 error generated.

In file included from /Users/kyle/frc-2019-beta/frc-2019-beta/src/main/cpp/commands/ExampleCommand.cpp:8:
In file included from /Users/kyle/frc-2019-beta/frc-2019-beta/src/main/include/commands/ExampleCommand.h:10:
In file included from /Users/kyle/.gradle/caches/transforms-2/files-2.1/6a64cf007cf067aa3f4c77f31619150a/wpilibc-cpp-2020.1.1-beta-1-headers/frc2/command/CommandBase.h:11:
In file included from /Users/kyle/.gradle/caches/transforms-2/files-2.1/6a64cf007cf067aa3f4c77f31619150a/wpilibc-cpp-2020.1.1-beta-1-headers/frc/smartdashboard/SendableHelper.h:13:
In file included from /Users/kyle/.gradle/caches/transforms-2/files-2.1/813d3f1bfc80aadce9047e7e0690a758/wpiutil-cpp-2020.1.1-beta-1-headers/wpi/Twine.h:14:
In file included from /Users/kyle/.gradle/caches/transforms-2/files-2.1/813d3f1bfc80aadce9047e7e0690a758/wpiutil-cpp-2020.1.1-beta-1-headers/wpi/StringRef.h:13:
/Users/kyle/.gradle/caches/transforms-2/files-2.1/813d3f1bfc80aadce9047e7e0690a758/wpiutil-cpp-2020.1.1-beta-1-headers/wpi/STLExtras.h:34:10: fatal error: 'optional' file not found
#include <optional>
         ^~~~~~~~~~
1 error generated.

In file included from /Users/kyle/frc-2019-beta/frc-2019-beta/src/main/cpp/Robot.cpp:8:
In file included from /Users/kyle/frc-2019-beta/frc-2019-beta/src/main/include/Robot.h:10:
In file included from /Users/kyle/.gradle/caches/transforms-2/files-2.1/6a64cf007cf067aa3f4c77f31619150a/wpilibc-cpp-2020.1.1-beta-1-headers/frc/TimedRobot.h:14:
In file included from /Users/kyle/.gradle/caches/transforms-2/files-2.1/6a64cf007cf067aa3f4c77f31619150a/wpilibc-cpp-2020.1.1-beta-1-headers/frc/ErrorBase.h:12:
In file included from /Users/kyle/.gradle/caches/transforms-2/files-2.1/813d3f1bfc80aadce9047e7e0690a758/wpiutil-cpp-2020.1.1-beta-1-headers/wpi/StringRef.h:13:
/Users/kyle/.gradle/caches/transforms-2/files-2.1/813d3f1bfc80aadce9047e7e0690a758/wpiutil-cpp-2020.1.1-beta-1-headers/wpi/STLExtras.h:34:10: fatal error: 'optional' file not found
#include <optional>
         ^~~~~~~~~~
1 error generated.

> Task :compileFrcUserProgramTestOsxx86-64DebugGoogleTestExeFrcUserProgramCpp FAILED
In file included from /Users/kyle/frc-2019-beta/frc-2019-beta/src/main/cpp/RobotContainer.cpp:8:
In file included from /Users/kyle/frc-2019-beta/frc-2019-beta/src/main/include/RobotContainer.h:10:
In file included from /Users/kyle/.gradle/caches/transforms-2/files-2.1/6a64cf007cf067aa3f4c77f31619150a/wpilibc-cpp-2020.1.1-beta-1-headers/frc2/command/Comma
nd.h:10:
In file included from /Users/kyle/.gradle/caches/transforms-2/files-2.1/6a64cf007cf067aa3f4c77f31619150a/wpilibc-cpp-2020.1.1-beta-1-headers/frc/ErrorBase.h:12:
In file included from /Users/kyle/.gradle/caches/transforms-2/files-2.1/813d3f1bfc80aadce9047e7e0690a758/wpiutil-cpp-2020.1.1-beta-1-headers/wpi/StringRef.h:13:
/Users/kyle/.gradle/caches/transforms-2/files-2.1/813d3f1bfc80aadce9047e7e0690a758/wpiutil-cpp-2020.1.1-beta-1-headers/wpi/STLExtras.h:34:10: fatal error: 'optional' file not found
#include <optional>
         ^~~~~~~~~~
1 error generated.

[... 140 lines cut ...]
> Multiple build operations failed.
      C++ compiler failed while compiling RobotContainer.cpp.
      C++ compiler failed while compiling ExampleCommand.cpp.
      C++ compiler failed while compiling ExampleSubsystem.cpp.
      C++ compiler failed while compiling Robot.cpp.
  See the complete log at: file:///Users/kyle/frc-2019-beta/frc-2019-beta/build/tmp/compileFrcUserProgramOsxx86-64ReleaseExecutableFrcUserProgramCpp/output.txt
   > C++ compiler failed while compiling RobotContainer.cpp.
   > C++ compiler failed while compiling ExampleCommand.cpp.
   > C++ compiler failed while compiling ExampleSubsystem.cpp.
   > C++ compiler failed while compiling Robot.cpp.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.
==============================================================================

* Get more help at https://help.gradle.org

BUILD FAILED in 3m 30s
8 actionable tasks: 4 executed, 4 up-to-date
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

OSX 10.13.5
WPILib Beta 1

PeterJohnson, 2019-10-21:
You need a more recent MacOS compiler, or disable desktop builds.

Kython89, 2019-11-09:
I ended up having to upgrade to OSX 10.14.6 in order to upgrade to XCode 11.2 but now the builds work with desktop support enabled.

There are quite a few errors but they don't appear to cause a problem.  Is this expected?

```
clang++: error: unable to read SDK settings for '/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk'
```

Full log
```
kyles-mbp:beta3 kyle$ ./gradlew build
xcrun: error: unable to read SDK settings for '/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk'
xcrun: error: unable to read SDK settings for '/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk'

> Task :compileFrcUserProgramOsxx86-64DebugExecutableFrcUserProgramCpp
Warnings in file RobotContainer.cpp ....
clang++: error: unable to read SDK settings for '/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk'
clang++: error: unable to read SDK settings for '/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk'
Warnings in file ExampleSubsystem.cpp ....
clang++: error: unable to read SDK settings for '/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk'
clang++: error: unable to read SDK settings for '/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk'
Warnings in file ExampleCommand.cpp ....
clang++: error: unable to read SDK settings for '/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk'
clang++: error: unable to read SDK settings for '/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk'
Warnings in file Robot.cpp ....
clang++: error: unable to read SDK settings for '/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk'
clang++: error: unable to read SDK settings for '/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk'

> Task :compileFrcUserProgramOsxx86-64ReleaseExecutableFrcUserProgramCpp
Warnings in file RobotContainer.cpp ....
clang++: error: unable to read SDK settings for '/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk'
clang++: error: unable to read SDK settings for '/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk'
Warnings in file ExampleCommand.cpp ....
clang++: error: unable to read SDK settings for '/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk'
clang++: error: unable to read SDK settings for '/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk'
Warnings in file Robot.cpp ....
clang++: error: unable to read SDK settings for '/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk'
clang++: error: unable to read SDK settings for '/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk'
Warnings in file ExampleSubsystem.cpp ....
clang++: error: unable to read SDK settings for '/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk'
clang++: error: unable to read SDK settings for '/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk'

> Task :runFrcUserProgramTestOsxx86-64DebugGoogleTestExe
HAL Extensions: No extensions found
[==========] Running 0 tests from 0 test suites.
[==========] 0 tests from 0 test suites ran. (0 ms total)
[  PASSED  ] 0 tests.

> Task :runFrcUserProgramTestOsxx86-64ReleaseGoogleTestExe
HAL Extensions: No extensions found
[==========] Running 0 tests from 0 test suites.
[==========] 0 tests from 0 test suites ran. (0 ms total)
[  PASSED  ] 0 tests.

BUILD SUCCESSFUL in 52s
18 actionable tasks: 18 executed
```

Kython89, 2019-11-09:
Actually running this clear everything up.
```
xcode-select --install
```

closed 2019-11-09 by Kython89