wpilibsuite/allwpilib #1978: Add computer vision pose estimation and latency compensation example

calcmogul, 2019-10-23:
In the spirit of incorporating state-of-the-art features into WPILib, we should add an example that demonstrates how to incorporate pose measurements from OpenCV's PnP or PnPransac into a pose estimate using a backlog of poses to compensate for camera latency. I've seen a few teams do this over the years, and it's an effective technique that has a lot of small things to get right.

We should also have an example vision pipeline that has the general algorithm for finding a target contour and using that with PnP. It should avoid code referencing a specific game target. This feature is similar to latency comp where it's a solved problem with a lot of tricky bits.

joshuathooyavan, 2021-12-26:
Would this use the PoseEstimator classes?

calcmogul, 2021-12-26:
We have pose estimator examples, but they don't show how to turn typical vision measurements like pitch and yaw into pose that can be given to the pose estimator.

Pursuant to that, both the WPILib devs and the PhotonVision devs would like to see https://github.com/PhotonVision/photonvision/tree/master/photon-lib incorporated into WPILib, but someone has to copy the files over and write sufficient unit tests (the existing lib has sparse Java tests and no C++ tests).

joshuathooyavan, 2022-09-12:
Can't PhotonLib be incorporated as a git submodule? Why did you mention "copy files over"?

calcmogul, 2022-09-12:
That comment is outdated. The relevant functions were added in https://github.com/wpilibsuite/allwpilib/pull/4124, so now someone just has to write examples.

PR #4901 (2023-01-19): [examples] Add Computer Vision Pose Estimation and Latency Compensation Example
This PR updates the existing differentialdriveposeestimator example to include computer vision pose estimation and latency compensation. 

The example generates a simulated cameraToTarget transformation, which is then fed into `ComputerVisionUtil.objectToRobotPose()` to compute the robot's field-relative position exclusively from vision measurements. The vision measurements are applied through ` DifferentialDrivePoseEstimator.addVisionMeasurement()`. 

Resolves #1978

The updated example constructs an `AprilTagFiel [... cut]
closed 2023-01-19 by PeterJohnson