wpilibsuite/allwpilib #1990: Create a new Class/Command menu imports old class templates

PeterJohnson, 2019-10-25:
[beta test report]

It seems that when you bring up the context menu and select to Create a new Class/Command the result imports the old templates from the [edu.wpi.first.wpilibj] and not the [edu.wpi.first.wpilibj2]

I'm not sure the best solution here for individuals that may hold on to the old way and those that will embrace the new. Maybe sub menus from the Create a new Class/Command > [2020 Class Model, 2019 Class Model] 

I think it will come down to if tools like RobotBuilder get caught up. Also I'm curios if a mix of these two libraries would still work? I suspect not.

ThadHouse, 2019-10-25:
The templates just need to be added to 

https://github.com/wpilibsuite/allwpilib/tree/master/wpilibcExamples/src/main/cpp/commands

and 

https://github.com/wpilibsuite/allwpilib/tree/master/wpilibjExamples/src/main/java/edu/wpi/first/wpilibj/commands

This is also where we would mark certain commands as old or new style.

Oblarg, 2019-10-25:
A mix will definitely not work.

sciencewhiz, 2020-01-01:
Fixed by wpilibsuite/allwpilib#2016

closed 2020-01-01 by sciencewhiz