wpilibsuite/allwpilib #2041: XBoxController input does not update during autonomous mode

agasser, 2019-11-06:
`XBoxController` input does not update during autonomous mode. Maybe this is intentional if 2020 will not allow teleop control during autonomous period. However, there is an issue if this is intentional because `XBoxController` seems to return cached values from the start of the game. Our 2019 autonomous runs a command to calibrate the manipulators, and then it runs the `DriveTrainSubsystem`'s default command to allow teleop control. With 2020 beta 2, the `XBoxController` returns the positions the controls were in when the game starts.

I was testing that teleop controls were disabled during our calibration, by attempting to drive forward at full speed with the XBox controller. When the calibration routine ended the robot took off at full speed (as expected because I was holding the XBox stick at full forward), but I was unable to stop it with the `XBoxController` because the values were cached. I would expect this to be 
a common paradigm - run an autonomous command that might be shorter than the duration of autonomous mode and use the Drivetrain subsystem's default command for teleop control.

TL;DL; If you are intentionally disabling `XBoxController` input during autonomous, you probably should return neutral positions for all of the controls instead of caching the values at robot start.

For reference, here is the code I was running: https://github.com/STMARobotics/frc-7028-2019/tree/f24d8a9f96744079ef69cd9a6623b96e1770ff61

ThadHouse, 2019-11-06:
This has actually been the behavior of netcomm for all years leading up to 2019. We actually don't have control over that behavior, which is handled at a level lower then us. In prior years, the latching current values was an expected behavior, as it let you use axes and buttons to set auto modes that could be read during auto.

I agree it is a bit of an issue for beta testing teams, but it was just how the image came out and is not something we can change or work around at the wpilib level.

agasser, 2019-11-06:
Interesting. I suppose a workaround in my case would be to not attach the teleop default command for a subsystem during autonomous - or the command will need to take the robot mode into account. If I don't implement that workaround and the driver's controller's axis is depressed when the robot starts it will run out of control after the auto command ends, until teleop mode starts.

It might be worth noting in the XBoxController (and related classes) that the input is not updated during autonomous.

coach-streeter, 2019-11-06:
Hmm.  Does this also mean that a team running 2020 beta code at a 2019 offseason event will have latched OI inputs during the sandstorm period?

PeterJohnson, 2019-11-06:
Yes

coach-streeter, 2019-11-06:
Hmm.  That’s unfortunate.  We have been planning to run the beta code and this coming Saturday’s event Battle of the Bay in NH, but that’s an unfortunate side effect.  Although we (1519) run full auto for sandstorm anyway, we do have the option of taking over operator control if something goes amiss, such as a robot collision with an alliance member...

agasser, 2019-11-06:
@coach-streeter, Right, that will not work.
1) You won't be able to take teleop control during the sandstorm because the code cannot read updated HID input.
2) You would need to make sure no teleop commands run during autonomous because the HID input cannot be trusted. For example, if your teleop command runs as soon as your autonomous command is done you cannot let your teleop command control the drivetrain with HID input. (This is how I ran into a cabinet in the lab)

AustinShalit, 2019-12-27:
Oops

agasser, 2020-01-11:
Looks like this was intentional since 2020 has auto

closed 2020-01-11 by agasser