wpilibsuite/allwpilib #2044: gyrodrivecommands example shares use of a subsystem across commands

agasser, 2019-11-06:
The example [gyrodrivecommands/RobotContainer](https://github.com/wpilibsuite/allwpilib/blob/cfe23c5cd038132098a995145f5fea3fa3bfd82a/wpilibjExamples/src/main/java/edu/wpi/first/wpilibj/examples/gyrodrivecommands/RobotContainer.java#L67) binds `RunCommand`s to `kBumperRight` without requiring `DriveSubsystem` on line 67.

This circumvents the `CommandScheduler`'s resource management and could cause concurrency issues. Yes, it's just an example, but it should demonstrate the proper way to use commands and subsystems

I was trying to think through how teleop control could work entirely with button binding, but this is a questionable example. If done properly by requiring the `DriveSubsystem`, the robot would temporarily lose control when switching in and out of slow mode because the default command would be interrupted to set the `DriveSubsystem`'s max output.

PeterJohnson, 2019-11-06:
I don’t see how it can cause concurrency issues, unless by that you mean conflict with another command?  The scheduler is all a single thread.  The other way to structure this would be to have a separate subsystem for just the speed scaling, which would avoid the interruption risk approach.

JefferMC, 2019-11-06:
I agree.  From a philosophical point of view,  buttons should invoke commands., commands should interact with the subsystems.  A button shouldn't directly make a subsystem method call.  I would tell my students to not do it this way, and an example program surely should demonstrate proper isolation of responsibilities.

agasser, 2019-11-06:
@PeterJohnson, as @JefferMC noted, it's not a good practice. Only commands should interact with subsystems. Commands declare their requirements and they should be guaranteed exclusive control of the subsystem.

To speak specifically to possible concurrency issues in this case, without knowing the internals `PIDController` claims to run on a separate thread. I would be concerned because `DifferentialDrive` is not threadsafe. If you call `DifferentialDrive.setMaxOutput()` while concurrently calling `arcadeDrive()` you could run into problems.
For example, consider [this code in `DifferentialDrive`](https://github.com/wpilibsuite/allwpilib/blob/cfe23c5cd038132098a995145f5fea3fa3bfd82a/wpilibj/src/main/java/edu/wpi/first/wpilibj/drive/DifferentialDrive.java#L225):
```
    m_leftMotor.set(MathUtils.clamp(leftMotorOutput, -1.0, 1.0) * m_maxOutput);
    double maxOutput = m_maxOutput * m_rightSideInvertMultiplier;
    m_rightMotor.set(MathUtils.clamp(rightMotorOutput, -1.0, 1.0) * maxOutput);
```
If the value of `m_maxOutput` changes in the middle you will get unpredictable results.

calcmogul, 2019-11-06:
wpilibj.controller.PIDController doesn't run in a separate thread.

agasser, 2019-11-06:
@calcmogul, You're right. I was looking at the deprecated [`PIDController`](https://github.com/wpilibsuite/allwpilib/blob/master/wpilibj/src/main/java/edu/wpi/first/wpilibj/PIDController.java).

Oblarg, 2019-11-06:
This is entirely safe, and is perfectly fine practice.  There's no reason that command should necessarily interrupt other commands running on the drive, so it doesn't need a requirement.

JefferMC, 2019-11-06:
Yes a **Command** doesn't _necessarily_ need to interrupt other commands, and need not _require_ the subsystem.  However, this isn't a **Command**.  I've always considered it poor form to bypass Commands and go straight to the Subsystem from the OI (like this example is doing).

Oblarg, 2019-11-06:
It is very much command - it's a shorthand for an InstantCommand.

JefferMC, 2019-11-06:
Oh?  A command gets created, scheduled, shows up in the Dashboard, etc.?  Or do you equate directly accessing a Subsystem with the term "InstantCommand?"

Oblarg, 2019-11-06:
`button.whenPressed(subsystem::foo)` is literally a wrapper on `button.whenPressed(new InstantCommand(subsystem::foo))`.  Saying it's "not a command" is flatly inaccurate.

JefferMC, 2019-11-06:
I cede the point.  To be honest, I haven't dug into the libraries to explore them for sometime.  However, I must say that extensions like this are quite easy to overlook (i.e. absent this wrapper, the code would compile and do exactly what I thought).

Oblarg, 2019-11-06:
Absent the wrapper, there would be no button binding method that takes a
runnable.

agasser, 2019-11-06:
> There's no reason that command should necessarily interrupt other commands running on the drive, so it doesn't need a requirement.

My concern is that it's not entirely up to that command to decide. It's fair to say _that_ command doesn't require exclusive access, but it's not fair to say _no other_ command require exclusive access. To use an analogy, that's like borrowing your roommates car without asking if they need it at the same time.

Ultimately, it might be "safe" in this case but it's not an **example** of what good looks like.

Oblarg, 2019-11-06:
The action of that command (halving the drive speed) is not an action one
would expect to interrupt the action of any other ordinary drive command,
though.  It's an action that *should* be generally safe.

Starlight220, 2020-04-19:
As I see it, the same discussion goes for `Subsystem.periodic()`.

closed 2021-11-09 by calcmogul