wpilibsuite/allwpilib #2049: Decorators that take Runnable should accept required subsystems

agasser, 2019-11-06:
The decorator methods on `Command` are great and make building command groups easy. I suggest adding requirements as a varargs argument to the decorator methods that take `Runnable` to help make it easy to follow best practices.

As an example, [`andThen`](https://github.com/wpilibsuite/allwpilib/blob/master/wpilibNewCommands/src/main/java/edu/wpi/first/wpilibj2/command/Command.java#L134) should be update like this:
```
  default SequentialCommandGroup andThen(Runnable toRun, SubSystem... requirements) {
    return new SequentialCommandGroup(this, new InstantCommand(toRun, requirements));
  }
```

Oblarg, 2019-11-06:
As these are vararg, there's literally no cost to adding this.

agasser, 2019-11-06:
If you make this change, I'd suggest also making a quick pass over the examples too.

For reference, [frisbeebot](
https://github.com/wpilibsuite/allwpilib/blob/master/wpilibjExamples/src/main/java/edu/wpi/first/wpilibj/examples/frisbeebot/RobotContainer.java#L53) could benefit from this addition. Notice at the end of this autonomous chain that `m_shooter` is not required when `m_shooter.disable()` and `m_shooter.stopFeeder()` are called.

```
  private final Command m_autoCommand =
      // Start the command by spinning up the shooter...
      new InstantCommand(m_shooter::enable, m_shooter).andThen(
          // Wait until the shooter is at speed before feeding the frisbees
          new WaitUntilCommand(m_shooter::atSetpoint),
          // Start running the feeder
          new InstantCommand(m_shooter::runFeeder, m_shooter),
          // Shoot for the specified time
          new WaitCommand(kAutoShootTimeSeconds))
          // Add a timeout (will end the command if, for instance, the shooter never gets up to
          // speed)
          .withTimeout(kAutoTimeoutSeconds)
          // When the command ends, turn off the shooter and the feeder
          .andThen(() -> {
            m_shooter.disable();
            m_shooter.stopFeeder();
          });
```

Oblarg, 2019-11-06:
On second thought, the C++ lib is not so fortunate, and would require overloads.  I suppose that's not *so* terrible, though.

agasser, 2019-11-06:
@Oblarg good call. I didn't realize the command group would inherit all of the requirements. That makes sense.

agasser, 2019-11-06:
As I look through the code more, I realized `Trigger` and its subclasses could benefit from this improvement as well. It would be helpful to allow requirements to be passed to `Trigger.whenActive(Runnable)` and `JoystickButton. whenPressed(Runnable)`. This addition could be used to quickly address #2044 .

Would you like me to create a separate issue for `Trigger`?

Oblarg, 2019-11-06:
Nah, I noticed that as well; just modify the title of this one, I'll get to
both eventually.

Oblarg, 2019-11-09:
Addressed by https://github.com/wpilibsuite/allwpilib/pull/2059

closed 2019-11-11 by AustinShalit