wpilibsuite/allwpilib #2073: Configure WPILib install directory with environment variable

Modelmat, 2019-11-12:
Currently, WPILib is installed to ~/wpilib/year and this is not easily changeable. I find this location rather annoying as it clutters up ~/ . 

However, if it were possible to change the directory using an environment variable such as WPILIB_HOME this would allow team members to easily configure the location of the install to somewhere like ~/.wpilib.

Furthermore, it would likely to be possible to install WPILib system wide if it were installed to somewhere like /opt/wpilib (although it would likely need to be writeable?).

ThadHouse, 2019-11-18:
A change like this has massive reaching affects throughout the entire WPILib ecosystem. 

To start with, this requires updating multiple WPILib projects to support this, including the installer, along with figuring out what to do on updates, or if the environment variable is not set.

Afterwards, vendors need to do the same thing, which makes both their installers and documentation more complicated.

Finally, and the largest reason, is debuggability. Currently, the location might be different per platform, but the location is fixed once you know your platform. This makes finding an issue much more difficult. We know from experience that users like to change things, and then forget they changed those things when asking for help when something doesn't work. So we go through hours of debugging, because we are assuming a fixed location. Debugging teams is already hard enough, and adding another variable exponentially raises cost for very little gain.

Modelmat, 2019-11-19:
Would a push for a default install directory other than in ~/ be allowed?
This is the reason why I asked for this; I do not like having to clutter up my ~/ directory; and being able to use another hidden location (or even one in say /opt/wpilib) would be much preferable (IMO).

ThadHouse, 2019-11-19:
Is there a common unix path that is available to all users without sudo? That is one of the requirements we have, that everything can be installed without root access.

calcmogul, 2019-11-19:
`~/.local/share/frc2019` would be the right place per-user, but there is no globally accessible directory.

Modelmat, 2019-11-19:
I would prefer `~/.local/share/wpilib/<year>` over the `~/wpilib` directory.
There is currently no globally accessible directory either so this isn't something that is a must, for a new location (I would think?)

ThadHouse, 2019-11-19:
It is definitely too late to change anything for 2020. I (and others) don't like `.` folders because the hidden property is actually very difficult to address with debugging, especially nested like that. So we might be able to bring this up again next summer, but I'm not sure how far it would get.

Modelmat, 2019-11-19:
Yeah, okay.
Should I create another issue tracking this?

ThadHouse, 2019-11-19:
No. We can reopen this one, but it definitely will not be looked at until next summer.

ThadHouse, 2020-07-10:
Because of the cancellation of the 2020 season, and our goal to make things as painless as possible, there is 0 chance anything will happen for 2021. Just FYI.

Starlight220, 2022-09-20:
More than two years later, do we have any intention of changing the install directory?
If not, I suggest closing this issue as a wontfix.

closed 2022-10-09 by PeterJohnson