wpilibsuite/allwpilib #2089: Loop overrun timeout should be configurable

Daltz333, 2019-11-16:
A team may use the console to output system debug messages when not wanting to use NetworkTables. The loop overrun is not that useful for Java projects considering even the smallest stuff causes a very small 1-3ms loop overrun. I'd like the ability to configure this timeout to a configurable variable. 

Use case, bumping loop overrun to 20ms over the robot loop that way I still now when I have a *serious* loop overrun and can still see debug information.

Starlight220, 2020-09-17:
Can't this be done using the `period` parameter in the `TimedRobot`/`IterativeRobotBase` ctors? 
Teams wishing to do this can do this in their `Robot.java`:
```java
public class Robot extends TimedRobot {
  public Robot() {
    super(customPeriod);
  }
  //robotInit, etc...
}
```
Or would this have unwanted side effects?
Alternatively, we can split it into a separate `loopTimeout` field with getter/setters that are basically delegated to the watchdog. Although we wouldn't want it to be changed in the middle of the loop (because the watchdog clears all epochs and resets when the timeout is changed). To avoid that we might run into some concurrency/caching issues. 
It can maybe be done similarly to how the `CommandScheduler` loop handles commands that are added in the middle of a loop.

calcmogul, 2021-07-11:
The loop overrun threshold is automatically set to whatever period TimedRobot's constructor took. It's 20ms by default. A robot consistently overrunning 20ms by any amount is a serious problem the team should address.

We rate-limit the message to only print once per second, so it shouldn't drown out periodic messages.

We don't plan on making the timeout configurable separately from the TimedRobot period.

closed 2021-07-11 by calcmogul