wpilibsuite/allwpilib #2093: A SequentialCommandGroup used as a deadline never finishes

josedolf-staller, 2019-11-17:
In the Beta 2 libraries, setting a SequentialCommandGroup as the deadline for a ParallelDeadlineGroup results in the SequentialCommandGroup's isFinished() never returning true (and thus the ParallelDeadlineGroup never finishing). However, setting a ParallelCommandGroup or a ParallelRaceGroup as the deadline works just fine, and the same SequentialCommandGroup's isFinished() returns true when called normally.

Attached is an example command group we tested in autonomous mode, if it helps.
[AutonTest.zip](https://github.com/wpilibsuite/allwpilib/files/3856231/AutonTest.zip)

Oblarg, 2019-11-18:
The sequential group's `isFinished` method does actually return true; it just returns false again after `end()` is called, so the second call is missed.  Easily fixable but with the deadlinegroup logic; I'll make a PR.

PeterJohnson, 2019-11-18:
Fixed by #2095.

closed 2019-11-18 by PeterJohnson

GregoryLi360, 2024-03-02:
This is still happening on v2024.3.1. When all commands in a SequentialCommandGroup are finished, the SequentialCommandGroup itself never finishes.

PeterJohnson, 2024-03-02:
Please open a new issue. I suspect the cause is different.