wpilibsuite/allwpilib #2111: WPIFormat does not format files in wpilibNewCommands

prateekma, 2019-11-21:
Added a `Test.cpp` file with the following contents to `wpilibNewCommands/src/main/native/cpp/frc2/command`:
```cpp
class Hello
{

};
```
Ran `wpiformat` in the project root directory.
`Test.cpp` was unchanged.

prateekma, 2019-11-21:
Deleting the `.styleguide` file inside `wpilibNewCommands` seemed to make it work. New output:
```cpp
/*----------------------------------------------------------------------------*/
/* Copyright (c) 2019 FIRST. All Rights Reserved.                             */
/* Open Source Software - may be modified and shared by FRC teams. The code   */
/* must be accompanied by the FIRST BSD license file in the root directory of */
/* the project.                                                               */
/*----------------------------------------------------------------------------*/

class Hello {};
```

calcmogul, 2019-11-21:
Child .styleguide files override everything from the parent, not on a section-by-section basis. In other words, they don't inherit sections from the parent. clang-format works the same way. The child .styleguide here should have everything in the wpilibc one with a different "includeOtherLibs" section.

PR #2113 (2019-11-21): Fix incomplete .styleguide
Also clean up other .styleguides.

Fixes #2111.
commit 9a806746 (2019-11-21): Fix incomplete .styleguide (#2113)

Also clean up other .styleguides.

Fixes #2111.
closed 2019-11-21 by PeterJohnson