wpilibsuite/allwpilib #2137: ./gradlew Build failing on macOS Mojave v10.14.6

alarruda, 2019-11-27:
Having an issue building Robot Code. Whenever we try, there is the following error message. We have had this on two Mac machines and have tried to delete the cache and *.lock files and there has been no fix.

Here is the error message:

`Executing task: ./gradlew build   <

Starting a Gradle Daemon, 1 stopped Daemon could not be reused, use --status for details

FAILURE: Build failed with an exception.

* What went wrong:
Could not open cp_settings remapped class cache for 8rb1s1liblrkleh2v8nlrh7el (/Users/aarruda/.gradle/caches/5.0/scripts-remapped/settings_8s27cs4ph5vn42jaswvpl3t6s/8rb1s1liblrkleh2v8nlrh7el/cp_settingsa9190c3ddc71e1450eb3f3e7bc6a334f).
> Could not open cp_settings generic class cache for settings file '/Users/aarruda/Desktop/FRC Projects/FRC2019/settings.gradle' (/Users/aarruda/.gradle/caches/5.0/scripts/8rb1s1liblrkleh2v8nlrh7el/cp_settings/cp_settingsa9190c3ddc71e1450eb3f3e7bc6a334f).
   > BUG! exception in phase 'semantic analysis' in source unit '_BuildScript_' Unsupported class file major version 57

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 1s
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.` 

Attempted fixes:
1. `find ~/.gradle -type f -name "*.lock" -delete`
2. `rm -rf ~/.gradle/caches`
3. Manually delete the cache files.

ThadHouse, 2019-11-27:
For WPILib Projects, only Java 11 is supported. Based on the report, its showing you're executing gradle and its finding Java 13. You need to make Java 11 be your primary java version. I'm not fully sure how to do this on mac, so you might have to look this up.

alarruda, 2019-11-27:
Thanks for the prompt response @ThadHouse! How did you know that it was Java 13?

My first attempt at reverting to Java 11 failed. I will comment with the fix when I find it.

ThadHouse, 2019-11-27:
`Unsupported class file major version 57`. Class file version 57 is Java 13. 55 is 11.

sciencewhiz, 2019-11-27:
> Unsupported class file major version 57

Class file version 57 corresponds to JDK 13.

alarruda, 2019-11-27:
Ok, so I ran `export JAVA_HOME=$(/usr/libexec/java_home -v 11)` in terminal which sets the version to "11.0.5"
`➜  ~ java -version
java version "11.0.5" 2019-10-15 LTS
Java(TM) SE Runtime Environment 18.9 (build 11.0.5+10-LTS)
Java HotSpot(TM) 64-Bit Server VM 18.9 (build 11.0.5+10-LTS, mixed mode)`

Unfortunately, I am still getting the build error.

ThadHouse, 2019-11-27:
Run `./gradlew --stop`, and then clear those caches, and retry.

KenwoodFox, 2019-11-27:
You can edit `java.home` in your `settings.json` in VSC, that might help force gradle to look there.

AustinShalit, 2020-03-21:
Closing this. Please reach out if you have more questions.

closed 2020-03-21 by AustinShalit