wpilibsuite/allwpilib #2178: Watchdog should use DS.reportWarning

PeterJohnson, 2019-12-16:
Currently, Watchdog uses multiple calls to System.out.format() to generate the output.  This can end up interleaved with other messages coming from DriverStation.reportWarning().  Watchdog should instead print to a string and make one call to reportWarning().

Starlight220, 2020-03-14:
On it!

Starlight220, 2020-04-15:
Again, Java wasn't fixed.

Starlight220, 2020-04-19:
@PeterJohnson - this wasn't fixed! You closed my PR (#2413) to fix this, claiming it was fixed in a different PR (#2379) that was merged, but it wasn't fixed!

AustinShalit, 2020-04-19:
This issue is still open which means, as you’ve mentioned, the issue still exists. If you would like to rebase your PR to resolve it please do so and leave a comment to reopen the PR.

PR #2413 (2020-05-21): [wpilib] Moved Watchdog output to reportWarning()
Watchdog now prints with one call to DriverStation.reportWarning() instead of multiple calls to System.out.format() that can be divided by other output.
Resolves #2178
closed 2020-05-21 by PeterJohnson