wpilibsuite/allwpilib #2181: Add documentation for common tasks to skip requiring the full build

ThadHouse, 2019-12-17:
Gradle does support incremental builds of only parts of the project, but not many people know them. We should have documentation (and added lifecycle tasks if neccessary) to make these things possible.

Please reply to this issue with common tasks you would like to do, so I can build the right list.

ThadHouse, 2019-12-17:
I think for hal, ntcore, cscore, and wpiutil I will add `testDesktopJava` and `testDesktopCpp` tasks

For wpilibc, I'll add `testDesktopCpp` and in wpilibj I'll add `testDesktopJava`

For examples, I'll add `compileDesktopExamples` in C++. In Java, I'll add `compileDesktopExamples`

calcmogul, 2019-12-17:
As a wpilibc/wpilibj developer, I build the unit tests and examples for wpilibc and wpilibj. Debug-only would suffice. The other tasks like simulation and myRobot aren't relevant for library-only feature work I do.

calcmogul, 2020-08-26:
Fixed by #2182.

closed 2020-08-26 by calcmogul