wpilibsuite/allwpilib #2185: Rename Encoder to QuadratureEncoder

calcmogul, 2019-12-20:
We should disambiguate Encoder from DutyCycleEncoder and AnalogEncoder. It only supports quadrature encoders rather than all three.

calcmogul, 2019-12-20:
We should use this opportunity to evaluate Counter's and Encoder's APIs and reorganize anything for which it makes sense.

Starlight220, 2020-03-14:
I will now rename the Java Encoder class and pull request

calcmogul, 2020-03-14:
When you do that, you need to make Encoder a deprecated wrapper around QuadratureEncoder so users have a year to migrate.

Starlight220, 2020-03-15:
I will do the java part now.

Starlight220, 2020-03-15:
Should I already remove the deprecated PID parts?

calcmogul, 2020-03-15:
I'm not sure yet. Depends on whether we remove the old PIDController for 2021. We'll need to discuss that internally, because I could see that going either way.

spacey-sooty, 2024-10-12:
> We should use this opportunity to evaluate Counter's and Encoder's APIs and reorganize anything for which it makes sense.

Could we remove or modify the behaviour of `getPeriod` and `setMaxPeriod` here? They exist on `Counter` but are deprecated in `Encoder`

calcmogul, 2024-10-13:
Depends on what the 2027 control system ends up being. Counter exposes getPeriod() because the CounterJNI and the HAL do. The HAL does because it exposes time between pulses for computing rate.

auscompgeek, 2026-11-02:
> When you do that, you need to make Encoder a deprecated wrapper around QuadratureEncoder so users have a year to migrate.

If we do this for the 2027 beta, we're okay with breaking that compatibility. We can add that migration to the project importer to help teams migrate.