wpilibsuite/allwpilib #2198: Add NavigateToCommand

AustinShalit, 2019-12-26:
I was looking at this code and was hoping there was a more idiomatic way to write the code.

```
  public Command getCargoCommand() {
    if (vision.totalCargo > 0) {
      Cargo cargo = vision.cargo[0];
      TrajectoryConfig config = new TrajectoryConfig(0.25, 1)
          // Add kinematics to ensure max speed is actually obeyed
          .setKinematics(Constants.DriveConstants.kDriveKinematics);

      // An example trajectory to follow. All units in meters.
      Trajectory cargoTrajectory = TrajectoryGenerator.generateTrajectory(
          List.of(new Pose2d(0, 0, new Rotation2d(0)),
                  new Pose2d(cargo.x_offset, cargo.distance, new Rotation2d(0))),
          // Pass config
          config);

      RamseteCommand ramseteCommand = new RamseteCommand(
          cargoTrajectory,
          drive::getPose, 
          new RamseteController(2, 0.7),
          Constants.DriveConstants.kFeedforward,
          Constants.DriveConstants.kDriveKinematics,
          drive::getWheelSpeeds,
          new PIDController(0, 0, 0),
          new PIDController(0, 0, 0),
          drive::tankDriveVolts,
          drive);
      return ramseteCommand.andThen(() -> drive.tankDriveVolts(0, 0));
    }
    return new WaitCommand(.1);
  }
```

Maybe we need to add a command that generates a trajectory and then runs that trajectory?

carbotaniuman, 2020-01-03:
That would either have a massively untenable argument list (RamseteCommand is already pushing it), or not provide much of a readability benefit.

AustinShalit, 2020-01-03:
It is less about readability and more about allowing for this style of command to be run from a button. I am trying to find an answer to the question: How do I write a command that will drive to a vision target goal and connect that command to a joystick button?

carbotaniuman, 2020-01-03:
You can write this same function that returns a Command, and use the lambda overload of button to call the function and schedule it.

Starlight220, 2020-04-15:
> a more idiomatic way

> You can write this same function that returns a Command, and use the lambda overload of button to call the function and schedule it.

Since all calls to the constructor will probably use the same parameters, maybe we should create a `RamseteConfig` class that stores most of the parameters (perhaps everything except the followed trajectory?) and a RamseteCommand ctor overload that accepts the trajectory, a `RamseteConfig` object, and maybe other stuff not stored in the `RamseteConfig` object, then deconstructs the config object into the currently-existing constructor. I can write a Java version, can someone else do C++?

AustinShalit, 2020-04-15:
The focus here is to have a mechanism that accepts a coordinate and drives there without the calling function needing to worry about generating the trajectory and then running the Ramsete command rather than the number of parameters on RamseteCommand. That refactor might be useful (I haven't thought hard about it though) but it is tangential to the issue I was bringing up.

Starlight220, 2020-04-15:
> calling function needing to worry about generating the trajectory and then running the Ramsete command

We can have a `RamseteCommand` ctor or something else that generates the trajectory, and then schedules the `RamseteCommand` (does pretty much everything in the code above), but it would have an insanely long list of arguments (problem 1) and would give zero adjustment room for teams. 
Most medium(plus)-teams would need more adjustment/control over this, and most low-level teams probably won't be doing path following. Therefore, there is little to no use of this.

Starlight220, 2020-04-15:
To sum up, path following is too complicated, and has too many parameters for it to be "one and done".

calcmogul, 2021-07-11:
The TrajectoryCommand class added in #3366 serves the purpose of NavigateToCommand.

Gold856, 2024-06-18:
OBE by #6494? (RamseteCommand is deprecated, and we've decided that users should use LTVUnicycleController and compose the command manually instead)

closed 2024-06-18 by calcmogul