wpilibsuite/allwpilib #2213: Add the ability to get SpeedControllers from within a SpeedControllerGroup

DarthWunka, 2020-01-01:
It would be nice to use the individual SpeedControllers within a group without having to create them outside, I would have made the pull request my self but I am not well versed in C++

calcmogul, 2020-01-01:
Creating the motor controllers first then passing them to the constructor of SpeedControllerGroup is the intended usage. Since they are passed to the constructor, you still have access to the original objects (unless you do `SpeedControllerGroup(new Victor(0), new Victor(1))`, but don't do that).

sciencewhiz, 2020-01-01:
What is the use case for using an individual SpeedController that is also in a SpeedControllerGroup?

DarthWunka, 2020-01-01:
> Creating the motor controllers first then passing them to the constructor of SpeedControllerGroup is the intended usage. Since they are passed to the constructor, you still have access to the original objects (unless you do `SpeedControllerGroup(new Victor(0), new Victor(1))`, but don't do that).

Yeah, I have done that now, but why is creating them and immediately adding them to the speed controller group such a bad idea?

DarthWunka, 2020-01-01:
ew

calcmogul, 2020-01-01:
I was just pointing out that if you want to access the objects outside of the group, don't allocate them in the ctor. It's a "doctor, it hurts when I do this" "Then don't do that" type of thing.

DarthWunka, 2020-01-01:
Okay I know I could do it the other way just as easily, it is just an organizational thing, Thanks

closed 2020-01-01 by DarthWunka