wpilibsuite/allwpilib #2215: Add overloads that don't take std::initializer_list

virtuald, 2020-01-01:
Dynamic bindings can't use std::initializer_list. Most places in wpilibc have it, with the following exceptions:

* HAL::CreateEnum (simdevice)
* frc2::Button::WhenPressed
* frc2::Button::WhileHeld (requirements overload)
* frc2::Button::WhenReleased (requirements overload)
* frc2::Trigger::WhenActive (requirements overload)
* frc2::Trigger::WhileActiveContinous (requirements overload)
* frc2::Trigger::WhenInactive  (requirements overload)
* frc2::Command::BeforeStarting
* frc2::Command::AndThen
* frc2::CommandScheduler::RegisterSubsystem
* frc2::CommandScheduler::UnregisterSubsystem
* frc2::FunctionalCommand constructor
* frc2::InstantCommand constructor
* frc2::MecanumControllerCommand constructor (2 of them)
* frc2::NotifierCommand constructor
* frc2::PIDCommand constructor (2 of them)
* frc2::ProfiledPIDCommand constructor (4 of them)
* frc2::RamseteCommand constructor (2 of them)
* frc2::RunCommand constructor
* frc2::StartEndCommand constructor
* frc2::SwerveControllerCommand constructor
* frc2::TrapezoidProfileCommand constructor

PeterJohnson, 2020-01-01:
SimDevice::CreateEnum has an ArrayRef overload already

virtuald, 2020-01-01:
Edited the list.

PR #2216 (2020-01-02): Add ArrayRef overloads to new command classes
Also default requirements to {} in all cases for consistency.

Fixes #2215.
closed 2020-01-02 by PeterJohnson