wpilibsuite/allwpilib #2217: Add option for whether InstantCommand can run when disabled

RobinsonZ, 2020-01-01:
There's a couple of use cases for creating an `InstantCommand` that runs when the robot is disabled (zeroing gyro, etc.). In the old framework you could call `setRunWhenDisabled()` after constructing the `InstantCommand`, but right now you need to create an anonymous subclass like this:
```java
InstantCommand zero = new InstantCommand(() -> foo.doBar()) {
      @Override
      public boolean runsWhenDisabled() {
        return true;
      }
    };
```
which is terrible. It would be great to have either a constructor option in `InstantCommand` (and possibly others such as `RunCommand`) or, better yet, a decorator to specify whether any `CommandBase` or similar can run when disabled (although I'm unsure if that could work with the structure).

AustinShalit, 2020-01-02:
I think the decorator way is probably what we want to do here.

RobinsonZ, 2020-01-02:
Agreed, I just wasn't sure if it was possible given that `Command` is an interface so it wouldn't just be a matter of setting a value.

Starlight220, 2020-03-15:
Use disabledInit() for these things, because if someone wrote a RunCommand/InstantCommand that runs while disabled and is supposed to move a motor, it will be confusing.

RobinsonZ, 2020-04-15:
At the moment the `RunCommand`/`InstantCommand` will just silently cancel itself anyway, so I feel like it's not any more or less confusing---not to mention that if you need to explicitly enable running when disabled it's not unreasonable to expect you to know that motors can't move when the robots disabled.

Also, `disabledInit()` doesn't address the two use cases I personally have for this functionality. The first case is running a command based on a trigger at some arbitrary time whenever the robot is on (for example, zeroing the gyro with a Shuffleboard button or controller button). I don't want it to run every time the robot's disabled, and I also want the driver to be able to zero the gyro in teleop. The second use case for this is to easily start up periodic loops, such as telemetry or odometry updates, without having to put something into `robotInit()` (which is an even worse idea nowadays with the recommended `RobotContainer`-based architecture).

Starlight220, 2020-04-15:
> you need to explicitly enable running when disabled

Doesn't this work on Run/InstantCommand?

closed 2022-06-24 by PeterJohnson