wpilibsuite/allwpilib #2219: Simulator GUI fails to launch on Ubuntu

dawonn, 2020-01-03:
WPILIB 2020.1.2 on Ubuntu 18.04 the simulator appears to run the robot code but the GUI fails to load. Not sure where to look for hints but this is the console output:

> HAL Extensions: Attempting to load: libhalsim_guid
> HAL Extensions: Trying modified name: libhalsim_guid.soHAL Extensions: Failed to load library
> 
> ********** Robot program starting **********
> Not loading CameraServerShared
> Default RobotInit() method... Override me!
> Default DisabledInit() method... Override me!
> Default DisabledPeriodic() method... Override me!
> Default RobotPeriodic() method... Override me!

drewwhis, 2020-01-03:
I'm seeing the same behavior on openSUSE.

PeterJohnson, 2020-01-04:
From the project directory, can you please run `cat build/gradlerio_simulateJava.sh` and then run `ldd <name>` where <name> is the value of HALSIM_EXTENSIONS in that shell script?  And reply with the output of both commands?

drewwhis, 2020-01-04:
@PeterJohnson the script didn't exist; running the gradlew command launched the sim and subsequently running the VS Code command works now.

sciencewhiz, 2020-01-04:
I had to install libvulkan1 for the gui to work

PeterJohnson, 2020-01-05:
Closing as we've made the documentation update regarding installation of libvulkan (wpilibsuite/frc-docs#427).  If you're still having issues, please comment and/or reopen.

closed 2020-01-05 by PeterJohnson