wpilibsuite/allwpilib #2223: (2020) enabling LiveWindow does not disable CommandScheduler

crussell52, 2020-01-04:
In the 2019 implementation, enabling LiveWindow explicitly disabled the command scheduler. Presumably because it is unhealthy the systems may compete with one another for control over devices.

In the 2020 implementation, the comments imply this should still be the case, but I see no logic to do so.
```
/**
 * Set the enabled state of LiveWindow. If it's being enabled, turn off the scheduler and remove
 * all the commands from the queue and enable all the components registered for LiveWindow. If
 * it's being disabled, stop all the registered components and reenable the scheduler...
 */ 
```

Because of there are two command scheduler implementations provided as library it makes sense that LiveWindow() can no longer do this.  It is not even reasonable for the robot base class(es) to manage this, for the same reason. However, it may surprise teams if both are active at the same time.

It might be worth highlighting this behavior change in the docs and/or providing instructions for adding code to `testInit()` that disables the command scheduler.

sciencewhiz, 2020-01-08:
The templates don't have calls to the scheduler in testPeriodic. While the comment isn't correct, It's probably unlikely that teams would have trouble with this, unless then manually added calls to the scheduler in testPeriodic.

crussell52, 2020-01-08:
>It's probably unlikely that teams would have trouble with this, unless then manually added calls to the scheduler in testPeriodic.

Worth pointing out that the docs guide teams to run the scheduler in `robotPeriodic()`, so the scheduler will run in test mode unless they do additional work to prevent it.

Reference: https://docs.wpilib.org/en/latest/docs/software/commandbased/structuring-command-based-project.html#robot

commit cb66bcca (2020-01-13): Add callback handlers for LiveWindow (#2053)

Fixes #2223
closed 2020-01-13 by PeterJohnson