wpilibsuite/allwpilib #2225: Build errors due to '../cscore/OPENCV_JAR_FILE-NOTFOUND'

monarrk, 2020-01-04:
I'm attempting to build allwpilib from scratch on Void Linux using CMake. My process is as follows:
```
cmake -DATOMIC="/usr/lib/libatomic.so.1" -GNinja ..
ninja
```

I get this full error:
```
ninja: error: '../cscore/OPENCV_JAR_FILE-NOTFOUND', needed by 'jar/cscore.jar', missing and no known rule to make it
```

note: a similar error is thrown for the same reason when using GNU Make. This is not an issue with ninja.
I believe this is an issue with some variable because of the NOTFOUND, but I'm no expert with using CMake so I'm not really sure.

Some system info:
uname -a: Linux bandwagonbox 4.18.17_1 #1 SMP PREEMPT Tue Nov 6 19:10:50 UTC 2018 x86_64 GNU/Linux
cmake version 3.16.2
ninja version 1.9.0

Thank you!

ThadHouse, 2020-01-04:
Your sisters OpenCV likely doesn't include Java support. You can completely disable Java support with -DWITHOUT_JAVA=ON in when configuring cmake

monarrk, 2020-01-08:
This worked, but I would like java support if possible. Would I need to build libopencv from source with java support turned on in this case?

ThadHouse, 2020-01-08:
Yes, you would.

closed 2020-03-10 by AustinShalit