wpilibsuite/allwpilib #2230: Only One Instance of AddressableLED can be Created

brad95411, 2020-01-04:
Trying to create more than one instance of AddressableLED in Java at least throws a HAL error. 

`ledsLeft = new AddressableLED(7);
 ledsRight = new AddressableLED(8);`

Produces the following error:

Unhandled exception: edu.wpi.first.hal.util.UncleanStatusException: Code: -1004. HAL: No available resources to allocate
Error at frc.robot.Robot.robotInit(Robot.java:22): Unhandled exception: edu.wpi.first.hal.util.UncleanStatusException: Code: -1004. HAL: No available resources to allocate
at edu.wpi.first.hal.AddressableLEDJNI.initialize(Native Method)
at edu.wpi.first.wpilibj.AddressableLED.<init>(AddressableLED.java:29)
at frc.robot.Robot.robotInit(Robot.java:22)

Where line 22 is the assignment of the "ledsRight" variable.

It's not clear in the documentation if we are only supposed to be able to create one instance of AddressableLED. At least I couldn't find anything in my travels. I would have thought if we could only create one instance there would be a singleton to get and set the pin for when it's retrieved.

PeterJohnson, 2020-01-04:
The current hardware only supports a single instance.  The hardware might in the future support more than one, which is why it's not a singleton.

brad95411, 2020-01-04:
Fair enough, I didn't see anything in the documentation (it may be my own inability to see), which is why I asked. Thanks much!

closed 2020-01-04 by brad95411