wpilibsuite/allwpilib #2245: Composition semantics odd with `Button`

crussell52, 2020-01-08:
A `Button` subclass benefits from methods that add semantics that make more sense for buttons (such as `whenPressed()`).  However, these semantics don't carry through when using the composition techniques described in the manual (`and()`, `or()`, etc).

Note, the documentation correctly avoids the button semantics in its composition example.

crussell52, 2020-01-08:
It seems like this should be solvable with generics, somehow... but I can see that causing some cascading changes making this improvement impractical for the 2020 season.

crussell52, 2020-01-17:
Found this noted in the docs. Since this is well-documented behavior it is a non-issue and should be considered an enhancement request.

Starlight220, 2020-04-13:
Is this still relevant? If so, I can work on Java, someone else should work on C++. Beforehand, the way to fix this should be decided on (to prevent C++/Java inconsistencies). The problem is that the CommandScheduler is the one handling button-bound commands.

calcmogul, 2020-08-27:
Running commands when two buttons are pressed at the same time, for example, is useful. I'm not sure how we'd do it because I'm unfamiliar with how commands are scheduled, but I think an attempt should be made.

Starlight220, 2020-08-27:
The main problem is that the button objects don't really do anything - they wrap a call to a DriverStation function.
Binding commands to buttons simply adds a runnable to CommandScheduler of `if(button is pressed) schedule the command`. 
The button object saves no state whatsoever, so you can't mess with a specific button/command binding. 
The trigger composition returns a Trigger object, so it has only `isActive()` and not the button-specific synonyms.

crussell52, 2020-09-03:
@Starlight220 

The `edu.wpi.first.wpilibj2.command.button.Button` class is a subclass to `Trigger`. It provides different method names that make more sense for working with buttons. For example, `whenPressed()` rather than `whenActive()`.

However, when you use `mybutton.and()`, your suddenly switch to the `Trigger` verbs, losing that advantage.

That said... Since I wrote the original ticket, I've started to feel like the entire trigger semantics and command binding is sort of backwards.  My gut (after another year of seasoning) says that:

* The primary subject should be the command rather than a condition
* The semantics should _always_ be based on truthiness rather than having separate semantics for buttons and other conditions(such as sensors).

That would give us something like this:

```java
myBinder = new CommandBinder(myCommand);  // Bind myCommand to conditions.
myBinder.whileTrue(myBooleanProvider)
```

For combining conditions, whether they are two buttons, two sensors, or a button and a sensor could be easily done using lambdas:

two Buttons
```java
myBinder.whileTrue(() -> myXbox.getAButton() && myXbox.getBButton());
```

Button and Sensor
```java
myBinder.onChangeToTrue(() -> myXbox.getAButton() && myLimitSwitch.get()); // button and sensor
```

Button OR sensor
```java
myBinder.onChangeToFalse(() -> myXbox.getAButton() || myLimitSwitch.get()); 
```

What's interesting about this interface is that you could easily make multiple ways to activate a single command. Here, `myCommand` activates if a limit switch goes false OR a gamepad button is pressed.  

```java
myBinder = new CommandBinder(myCommand);

myBinder.onChangeToFalse(() -> myLimitSwitch.get())
myBinder.onChangeToTrue(() -> a.pressed())
```

Of course, that last example is a less common use case but we get strong and consistent semantics for the full range of simple use cases to complex use cases.

Any thoughts on this?

Starlight220, 2020-09-17:
Would renaming `Trigger.*Active()` to `Trigger.*True()` help?
All of the examples you brought above can be easily done with the Trigger API. For example, your last example can be done as:
```java
Trigger limitSwitchTrigger = new Trigger(limSwitch::get);
button.or(limitSwitchTrigger.not()).whenActive(myCommand);
```

closed 2021-11-12 by calcmogul