wpilibsuite/allwpilib #2246: `Sendable` functionality broken for both `Scheduler` and `CommandScheduler`

Oblarg, 2020-01-08:
`Scheduler` (old command framework) shows up on dashboard as an empty item with no state.  `CommandScheduler` shows up with `ids` and `names` arrays which are both empty regardless of what's actually scheduled, and without any `Cancel` array (which should be there).

Neither are able to be displayed as any Shuffleboard widget other than `NetworkTableTree`.

carbotaniuman, 2020-01-09:
The reason they don't display nicely is that they were never added as a widget.

https://github.com/wpilibsuite/shuffleboard/tree/master/plugins/base/src/main/java/edu/wpi/first/shuffleboard/plugin/base/widget

If added, what information should be available?

Oblarg, 2020-01-10:
It is definitely true that there is no widget, but neither class is working correctly regardless.

The way it's ostensibly supposed to work is by showing a list of currently-scheduled commands along with corresponding cancel buttons for each one.  Pressing the cancel button should add that commit's hash to the "cancel" array, which will cause it to be canceled in the update callback (after which the array is cleared).

I tried to write a unit test to figure out what was going wrong with `CommandScheduler`'s callback, but had trouble getting NetworkTables to work during testing.

If we're going to re-design these from the ground-up (since no widget actually exists), I think the aforementioned functionality *plus* a list of all registered subsystems and the commands currently requiring those subsystems would be helpful.

Starlight220, 2020-04-19:
What's up with this?

calcmogul, 2020-08-27:
I made an issue in the Shuffleboard repo for the widget part. The robot side is done, meaning this issue can be closed.

closed 2020-08-27 by calcmogul