wpilibsuite/allwpilib #2249: opencv_java344 could not be loaded from path or an embedded resource

gerth2, 2020-01-10:
Upon deploying to the roboRIO, the following java error is showing up in the rioLog:

```
java.io.IOException: opencv_java344 could not be loaded from path or an embedded resource.
attempted to load for platform /linux/athena/
at edu.wpi.first.wpiutil.RuntimeLoader.loadLibraryHashed(RuntimeLoader.java:125)
at edu.wpi.cscore.CameraServerJNI.<clinit>(CameraServerJNI.java:40)
at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:213)
at frc.robot.Main.main(Main.java:27)
```

A blank, brand new project deploys and runs fine on the same roboRIO. We've attempted re-imaging, but not seeing that change the behavior.

Is this a known issue? And/or is there a good place to start debugging this? I'm working through our history to see if I can narrow down where the code stopped working.

Code is here: https://github.com/RobotCasserole1736/RobotCasserole2020 Repo is private, but I can add anyone who wants to take a peek.

calcmogul, 2020-01-10:
That version of OpenCV is from 2019. The 2020 version is 3.4.7.

gerth2, 2020-01-10:
So this was created from the 2020 wpilib install. We've added a few dependencies to build.gradle, but haven't altered the build infrastructure much more. 

Where is the GradleRIO project's library year controlled from?

gerth2, 2020-01-10:
Update: Found the commit where it started happening. It was where we added our ADIS16448 Gyro. When we added the vendor libraries, it created the following file:

```
{
    "fileName": "ADIS16448.json",
    "name": "ADIS16448-IMU",
    "version": "2019-r3",
    "uuid": "38c21ab6-aa8b-44bc-b844-8086c77f09ec",
    "mavenUrls": [
        "http://maven.highcurrent.io/maven"
    ],
    "jsonUrl": "http://maven.highcurrent.io/vendordeps/ADIS16448.json",
    "javaDependencies": [
        {
            "groupId": "com.github.juchong",
            "artifactId": "adis16448-java",
            "version": "2019-r3"
        }
    ],
    "jniDependencies": [],
    "cppDependencies": [
        {
            "groupId": "com.github.juchong",
            "artifactId": "adis16448-cpp",
            "version": "2019-r3",
            "libName": "libadis16448imu",
            "headerClassifier": "headers",
            "sharedLibrary": false,
            "skipInvalidPlatforms": true,
            "binaryPlatforms": [
                "linuxathena"
            ]
        }
    ]
}
```

I'm noticing a handful of references to 2019 here.... also having a vague recollection of a CD post where all this was discussed....

gerth2, 2020-01-10:
Bingo: https://www.chiefdelphi.com/t/adis16448-opencv-java344-dependency-missing/370625

Now to go bug Analog Devices for their ETA on a new release...

closed 2020-01-10 by gerth2