wpilibsuite/allwpilib #2286: Roborio Cannot Deploy [JSchException]

WuYuanhun, 2020-01-17:
We currently encounter a problem that we cannot deploy code to our robot due to missing target.
However, we actually can ping to roborio and connect to roborio using sftp tool.
We believe that it is somehow related to JSchException. 
PS. We have tried to re-image both our roborio and radio and test on both MacOS and Windows 10.
We are now under VSCode-WPILib 2020.2.2.
Thanks !

`

> Task :discoverRoborio
Discovering Target roborio
admin @ roborio-5308-FRC.local: Resolved but not connected.
  Reason: JSchException
  connection is closed by foreign host
admin @ 10.53.8.2: Resolved but not connected.
  Reason: JSchException
  connection is closed by foreign host
admin @ 172.22.11.2: Resolved but not connected.
  Reason: JSchException
  connection is closed by foreign host
admin @ roborio-5308-FRC: Failed resolution.
  Reason: UnknownHostException
  roborio-5308-FRC: nodename nor servname provided, or not known
admin @ roborio-5308-FRC.lan: Failed resolution.
  Reason: UnknownHostException
  roborio-5308-FRC.lan: nodename nor servname provided, or not known
admin @ roborio-5308-FRC.frc-field.local: Failed resolution.
  Reason: TimeoutException
  Discovery timed out.
1 other action(s) not started.
Run with --info for more details

> Task :discoverRoborio FAILED
Missing Target!
=============================================
Are you connected to the robot, and is it on?
=============================================
GradleRIO detected this build failed due to not being able to find "roborio"!
Scroll up in this error log for more information.

FAILURE: Build failed with an exception.`

Kevin-OConnor, 2020-01-17:
Do you have a password set on your roboRIO?

WuYuanhun, 2020-01-18:
> Do you have a password set on your roboRIO?

Nope. We did not set password.

sciencewhiz, 2020-01-18:
Is the firewall enabled on the radio?

WuYuanhun, 2020-01-18:
> Is the firewall enabled on the radio?

The firewall is disabled.

WuYuanhun, 2020-01-18:
Update: We have successfully connect to the roborio and deploy code on Windows 10 laptop.
**The issue only occurs on MacOS now.**

WuYuanhun, 2020-01-18:
Next step, we plan to test on another MBP to see whether this problem would occur.

closed 2020-01-27 by WuYuanhun