wpilibsuite/allwpilib #2296: Trajectory Generator violating constraints

AustinShalit, 2020-01-21:
It looks like the Trajectory Generator is violating wheel speed constraints.

Here is a graph of left and right wheel speeds vs time. The blue is generated and orange is how the robot performed. I am pretty sure the TrajectoryConfig being used has a max velocity of 2.7m/s – way slower than the ~4.5m/s outputted by the trajectory.

[image]

I used `vl=v-rω` and `vr=v+rω` to calculate left and right wheel speeds from the trajectory. The robot wheel speeds are from the encoders on the robot.

Code to reproduce the trajectory:
```
  @Override
  public void robotInit() {
    var kinematics = new DifferentialDriveKinematics(0.6889);
    var voltageConstraint = new DifferentialDriveVoltageConstraint(
        new SimpleMotorFeedforward(1.34, 3.13, 0.926),
        kinematics,
        10);
    TrajectoryConfig config = new TrajectoryConfig(2.7432, 2.0);
    config.setKinematics(kinematics);
    config.addConstraint(voltageConstraint);
    Trajectory grabTrajectory = TrajectoryGenerator.generateTrajectory(
        List.of(
            new Pose2d(0, 0, new Rotation2d(0)),
            new Pose2d(1, 0, new Rotation2d(3.14/2))),
            config);
    for (var s : grabTrajectory.getStates()) {
      System.out.println(s);
    }
  }
```

calcmogul, 2020-01-21:
Based on the SimpleFeedforward arguments, there is a max wheel speed constraint of `12 V / 3.13 V/m/s = 3.83 m/s`.

prateekma, 2020-01-21:
```java
@Test
  void test() {
    var kinematics = new DifferentialDriveKinematics(0.6889);
    var voltageConstraint = new DifferentialDriveVoltageConstraint(
        new SimpleMotorFeedforward(1.34, 3.13, 0.926),
        kinematics,
        10);
    TrajectoryConfig config = new TrajectoryConfig(2.7432, 2.0);
    config.setKinematics(kinematics);
    config.addConstraint(voltageConstraint);
    Trajectory grabTrajectory = TrajectoryGenerator.generateTrajectory(
        List.of(
            new Pose2d(0, 0, new Rotation2d(0)),
            new Pose2d(1, 0, new Rotation2d(3.14 / 2))),
        config);

    for (var state : grabTrajectory.getStates()) {
      var wheelSpeeds = kinematics.toWheelSpeeds(new ChassisSpeeds(
          state.velocityMetersPerSecond, 0, state.velocityMetersPerSecond * state.curvatureRadPerMeter));

      Assertions.assertTrue(wheelSpeeds.leftMetersPerSecond < 2.8);
      Assertions.assertTrue(wheelSpeeds.rightMetersPerSecond < 2.8);
    }
  }
```

All assertions pass for me.

prateekma, 2020-01-21:
[image]
Here are plots of left and right velocity vs. time. This confirms that no wheel speed goes above ~2.8 m/s. Seeing as your peaks are approximately double that of the peaks in this image, I am suspecting that you used track width instead of radius (which is track width / 2) in your calculations.

GrantPerkins, 2020-01-21:
Where is radius supposed to be used? I've seen nothing that takes that as a parameter.

AustinShalit, 2020-01-21:
Ah... I was using chassis velocity instead of just forward velocity.

closed 2020-01-21 by AustinShalit