wpilibsuite/allwpilib #2300: Write a step-by-step guide to making a JNI

calcmogul, 2020-01-22:
All the motor controller vendors use them, and vendor libs that need performance (like state-space) or want code deduplication require one. The process isn't documented and is error-prone. (I've been told it's easy, but my last attempt to make one for Drake went badly. The person who attempted after me didn't fare any better.)

PeterJohnson, 2020-01-22:
The build part or the source code part?

calcmogul, 2020-01-22:
The steps you have to go through starting from the vendor lib template to make a JNI to a C++ version of the library. @ThadHouse made a checklist for this on Discord this morning.

calcmogul, 2020-01-22:
We copied the Java native files and C++ JNI function impls out of wpiutil because there's no documentation on where any of the files come from or how to get them (I now know there's a Gradle command for it, but it's still undocumented functionality.)