wpilibsuite/allwpilib #2302: Unexpected PerpetualCommand behavior

kasemir, 2020-01-23:
The new fluent command API is nice, but doesn't always do what one might expect.
This looks like it should keep printing 'Hello':
```
Command demo1 = new PrintCommand("Hello").perpetually();
```

In fact it only prints 'Hello' once because `PrintCommand` only prints during `initialize()`.

This looks like it should keep printing 'Hello' and then 'Bye':
```
Command demo2 = new PrintCommand("Hello").andThen(new PrintCommand("Bye!")).perpetually();
```
Again it only prints that once for the reason mentioned above.
Worse, it ends in
```
IndexOutOfBoundsException: Index 2 out of bounds for length 2
        at java.base/jdk.internal.util.Preconditions.outOfBounds(Preconditions.java:64)
        at java.base/jdk.internal.util.Preconditions.outOfBoundsCheckIndex(Preconditions.java:70)
        at java.base/jdk.internal.util.Preconditions.checkIndex(Preconditions.java:248)
        at java.base/java.util.Objects.checkIndex(Objects.java:372)
        at java.base/java.util.ArrayList.get(ArrayList.java:458)
        at edu.wpi.first.wpilibj2.command.SequentialCommandGroup.execute(SequentialCommandGroup.java:66)
        at edu.wpi.first.wpilibj2.command.PerpetualCommand.execute(PerpetualCommand.java:44)
```

The issue is related to the `PerpetualCommand` masking  `isFinished()` and continuing to call `execute()`, instead of repeating the complete initialize, execute, end cycle.

At first glance, both issues could be fixed if the perpetual command worked more like this:
```
  public void initialize() {
    m_command.initialize();
  }

  @Override
  public void execute() {
    m_command.execute();
  }

  @Override
  public boolean isFinished() {
    if (m_command.isFinished()) {
        // Wrapped command ended?
        // Complete it, then re-initialize so we can execute next time around
        m_command.end(false);
        m_command.initialize();
    }
    return false;
  }  
```

Probably a bad idea to 'fix' this right now in the FIRST season, and there are likely other side effects from calling code that _does_ something from `isFinished`.
Still, the current 'perpetually' functionality is quite unexpected.

Starlight220, 2020-03-16:
This undermines the purpose of initialize(). 
There should be a solution for running InstantCommands perpetually, probably by refactoring InstantCommand. Either way, it is not something that should change in the middle of the season.

Starlight220, 2022-04-21:
This can be closed:
RepeatCommand (added in https://github.com/wpilibsuite/allwpilib/pull/4009) solves this, and https://github.com/wpilibsuite/allwpilib/issues/4176 would clean up any naming confusion.

kasemir, 2022-04-21:
Indeed, thanks!

closed 2022-04-21 by kasemir