wpilibsuite/allwpilib #2308: javadoc links to jdk classes are broken

sciencewhiz, 2020-01-24:
Clicking on the File link  at https://first.wpi.edu/FRC/roborio/release/docs/java/edu/wpi/first/wpilibj/Filesystem.html#getDeployDirectory() leads to a 404 error. It looks like Oracle changed the directory structure. We link to https://docs.oracle.com/en/java/javase/11/docs/api/java/io/File.html?is-external=true. The new link is https://docs.oracle.com/en/java/javase/11/docs/api/java.base/java/io/File.html

The link is controlled through https://github.com/wpilibsuite/allwpilib/blob/master/docs/build.gradle#L105

sciencewhiz, 2020-01-28:
updating the link to `https://docs.oracle.com/en/java/javase/11/docs/api/java.base` produces an error `javadoc: error - Error fetching URL: https://docs.oracle.com/en/java/javase/11/d ocs/api/java.base/` 

This may be fixed in jdk 12.0.1: https://bugs.openjdk.java.net/browse/JDK-8215291

PeterJohnson, 2020-03-29:
This appears to be fixed in 11.0.2.  Testing with 11.0.6 with the following (in build.gradle's subprojects block) generates correct links:
```
    tasks.withType(Javadoc) {
        options {
            links 'https://docs.oracle.com/en/java/javase/11/docs/api/'
        }
    }
```

PR #2775 (2020-10-16): [build] Split documentation into its own job
This also allows us to update the version of java used to build documentation.

Closes #2308
Closes #2177
closed 2020-10-16 by PeterJohnson