wpilibsuite/allwpilib #2315: new PIDController doesn't support enabled

sciencewhiz, 2020-01-28:
The PIDController widget for SmartDashboard/Shuffleboard has an enabled boolean property that isn't supported by the new PIDController class. While it was very important to be able to disable the old PIDController since it automatically wrote to the motor, it is still helpful to pass the enabled flag to user code for while tuning with LiveWindow.

cttdev, 2020-01-31:
With the new PIDController, the user is left with calling the `calculate()` method in their robot loop. I think it should be left to the user to implement an enable/disable toggle for the controller(perhaps using a SmartDashboard boolean) as they are the ones calling it, transitively they are the ones declaring when the output will be written too. This would also allow for the user to customize what the happens to the output when the controller is enable/disabled.

Starlight220, 2020-04-06:
If this is settled, this issue should be closed.

calcmogul, 2020-08-27:
We could add an example of the SmartDashboard boolean thing to frc-docs.

It's worth noting motors automatically disable in disabledPeriodic(). Users could just disable when changing the PID constants then reenable. Integrators would still run if the controller is in `robotPeriodic()`, but that was an issue with the old PIDController too.

Daltz333, 2020-09-20:
Transferring to frc-docs. Should document how users can implement this.