wpilibsuite/allwpilib #2323: Optimization of RobotBase class.

CircuitCraft42, 2020-01-31:
There is a ReentrantLock m_runMutex being used to
synchronize access to two fields: m_robotCopy and
m_suppressExitWarning. It would be more efficient
to make them volatile fields instead.

ThadHouse, 2020-01-31:
Those variables are only used during initialization and cleanup, and are not used during main robot code execution. That's definitely approaching premature optimization, plus volatile variables in Java have some weird semantics which a lock doesn't have, and would make the code more confusing. So this isn't something that would happen.

closed 2020-01-31 by ThadHouse

CircuitCraft42, 2020-02-01:
I agree that it would be premature optimization. I just began looking at the codebase and noticed that possibility almost immediately. Since assignment and retrieval of references or 32-bit or less fields is atomic, the only reason I could see for the lock was synchronization, and volatile fields ensure that that property would remain. I do understand, however, that it would be more difficult to read and understand the code.